spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #4: BOM vs Populating reference and Partslist

metapulp, 2018-03-27:
Hi, I'm ordering from Elecrow and I have these issues:

1: The spreadsheet with part quantities does not match the quantities on the populating list. So basically from the quantity list they think they have more soldering to do than the populating list and pictures of the board show.
2: Elecrow is asking for a partlist generated from Autodesk Eagle - which I don't know how to do with the Gerber files. Do you have a DXF? Can you generate that partlist? Thank you.

spenceraxani, 2018-03-28:
Hi,

I’ve added an extra unit to the quantity in the purchasing list, since I noticed some people were just buying single resistors and then losing them. 

The list to populate the boards is found in folder PCB_Files/SMT_reference. However, I do not have the BOM list for them to populate the boards. We use this project to teach students how to solder surface mount components.

The Gerber files (located here: PCB_Files/PCB_Gerber_files) contain the information required for Elecrow to simply make the PCB, not populate it.

Thanks,

Spencer

metapulp, 2018-03-28:
I understand. I’m also hiring Elecrow to populate most of the basic parts. Any chance you have the board(s) as a DXF and you could post it on github so I can generate the partlist file they are requesting? I’m not having any luck importing the Gerber in Eagle. Thanks!

metapulp, 2018-03-29:
Let me amend my comment. I'm trying to import files into Eagle with no luck, so I'm not seeing how to generate the partlist for Elecrow...

They are looking for something like this:
[image]

spenceraxani, 2018-03-29:
Hi,

I hope to generate the list at some point, but am unable to work on this at this time. I would recommend populating the components by hand for the moment being. 

Thanks,

Spencer