spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #6: HV to ground voltage measuring 33.6 volts. Recheck components in DC booster circuit. Is that acceptable since it exceeds the nominal 29.5 volts?

pedelen, 2018-05-16:


pedelen, 2018-05-16:
That was a great help. Loose connection on 226K resistor, now exactly 29.6 V output

pedelen, 2018-05-24:
Connected SiPM to 29.64  volts, but no counts. Arduino nano code rechecked. Don't have an oscilloscope to check test points. Checked SiPM resistance and is 10.5 megohms in one direction and infinite in the other. Should there  be a high count due to ambient photons? So if 0 count, maybe I should suspect the amplifier circuit?

spenceraxani, 2018-05-24:
Hi,

The most common problem I see is a missed or poor connection — particularly on the op-amp (90% of the time, it’s the opamp). I would start by adding new solder to those legs just to make sure. If you are able to upload code to the Arduino, I assume it’s probably working fine. You can also go through the other components on the board. I’ve also seen people mount the SiPM rotated (pin1 and 3 interchanged). If all this still doesn’t work, you’ll have to find an oscilloscope to nail down what part of the circuit is giving you problems.

You probably shouldn’t expose the SiPM to ambient light, but if you do, you still won’t see any counts. I think SiPMs have their own peculiar behavior with large amounts of light, but the circuit is designed to only be able to count below 1000Hz. We stretch to pulses to roughly one millisecond, and the Arduino will wait for a pulse to subside before looking for the next pulse. 

Thanks,

Spencer

> On May 23, 2018, at 7:15 PM, pedelen <notifications@github.com> wrote:
> 
>

pedelen, 2018-06-09:
Trying to track down problem. Each reset results in a count of 1 and then timer keeps going but no additional counts. Tried reloading sketch multiple times. Tried replacing SiPM and repopulating main PCB.

mmalluck, 2018-07-02:
Hello Pedelen,

Did you get your issue sorted? I have something similar happening with both of my V1 boards. 1 count and then nothing. If I leave them plugged in for a while, eventually they start counting. I found raising the "RESET_THRESHOLD" in the software to about 15 gets it counting immediately or just letting it warm-up for a 10 minutes and then resetting it.

My guess is there's some noise coupling into the circuit that settles out once everything warms up. USB power source doesn't seem to make a difference. If I can get some time in-front of an o'scope, I'll see if I can find the source.

pedelen, 2018-07-02:
Hello mmalluck,
Thank you for your help Spencer. Two v1 boards now working perfectly with original threshold settings. I am not sure what did the trick. I installed the standoffs, rewrapped the scintillator and used better optical coupling compound. Do the standoffs have any electrical function? The counts are now much closer to expected with better optical coupling (Rate ~ .55 cps).

mmalluck, 2018-07-02:
Do you mind me asking what you're using for optical coupling gel then vs now?

I'm currently using some silicone based '[scintilator coupling compound'](https://m.ebay.com/itm/Silicone-Optical-Coupling-Compound-for-PMT-Photomultiplier-Scintillator-Detector-/261918290249&ved=2ahUKEwiCkfCuh4HcAhUQvVkKHSSuCccQFjAAegQIBBAB&usg=AOvVaw2Nib31XFFvkr7g1rLa5iSK)off ebay and I'm wondering if it's contributing to this.

I'm looking at using some fiber optic gel instead.

spenceraxani, 2018-07-02:
Another potential source that could cause this problem is if you have a light leak. I typically wrap it in reflective foil, then put about 3 full layers of black electrical tape to make sure it’s isolated.

pedelen, 2018-07-02:
Hello mmalluck,
That is the exact same optical coupling compound that I am using now with good results. It appears that my issue was due to a light leak and it seems to be resolved now. I now realize I have been posting on the v2 thread about my issue with v1 boards.

closed 2018-07-02 by pedelen