spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #8: Can't find Cap C6 - 20nF 50V

rogerio-acioli, 2018-06-17:
Hi,

I'm having problems to find the C6 capacitor (20nF - 50V) to buy at my country. Could I have some problem in the funcionallity of the detector if I use a 22nF capacitor instead the nominal 20nF?
Thanks!

spenceraxani, 2018-06-17:
Hi,

Sometimes capacitors are labelled in pF (20,000 pF) rather than nF which could be why you are having trouble finding the correct value. If you need to use the 22nF capacitor, I don’t anticipate you running into a problem. That capacitor determines the charge and decay rate of the peak detector pulse. If you swap it for 22nF, it will charge a bit slower and decay over a slightly longer time. This means that the calculated SiPM pulse amplitude will also slightly change (probably at the 10% level).

Thanks,

Spencer

samr28, 2018-06-17:
I also had a few problems with some of the parts being out of stock. The filter features on Digikey are super helpful. I was able to find replacements for all of the ones that were either out of stock or no longer in production by just entering the values into the filter. So for that one, I ended up purchasing 478-10429-1-ND.

From what I understand, the main things that you need to look at is the capacitance (20 nF), the size (0805 and overall dimensions should be similar to original). You should be able to go with a higher voltage without any problem as that value is just the voltage that should not be exceeded.

Hope this helps!

rogerio-acioli, 2018-06-19:
Thansk for the answers, guys!
Fortunatelly I finally found the capacitor here, but the informations that you sent are very important though.

Thanks,
Rogério.