spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #9: Stand-offs Too Short / Screws too long

mmalluck, 2018-06-19:
The Part List specifies 3/8" stand-offs for SiPM PCB. The interfacing screws are 1/4" in length. Using two of these screws results in a clash as both screws come in contact inside the stand-off.  The SiPM also droops. This can be corrected by using a slightly longer standoff.

spenceraxani, 2018-06-21:
Hi,

Thanks for the catch. I was using the 3/8” standoffs prior to this without trouble since there is also two 1.6mm PCBs in-between the standoffs and the 1/4” screws as well as the fact that they are tapered. Never-the-less a 7/16” standoff keeps the scintillator a bit more level and fits nicely so I’ve updated the GitHub purchasing list.

What do you mean by the ’SiPM droops’. Do you mean you are looking at the signal coupling? Droop is a rather small effect at the low intensity signals that we are looking at, but there is footprints on the PCB for AC coupling if you are concerned with the signal droop. 

Thanks,

Spencer

spenceraxani, 2018-06-21:
> Hello Spencer,
>
> 'Droop' was purely in regards to the SiPM PCB not being level with the rest
> of the assembly.
>
> I'm almost done with my V1 versions and already seeing about building V2.
> The V1 SiPM PCBs should be compatible with V2, correct? I'm hoping I don't
> need to buy more SiPMs.
>
> Thanks,
> Michael Malluck

Hi,

I see what you mean. There are spacers that you can purchase from McMaster if you want to make it more level — or going to 7/16” brings it really close.

The SiPM assembly is fully compatible with the newer version of the detector.

Thanks,

Spencer