spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #11: Employing a regulated 5v source

mmalluck, 2018-06-29:
Just a suggestion for V3 hardware: As the quality of power delivered via USB can vary substantially between different sources (5.25v to 4.4v allowable), I would recommend using the Nano's 5v regulator to source 5v to the rest of the board. The on-board regulator is capable of sourcing 800ma @ 5v, so this should be plenty for the other equipment. By doing this it should help cut down on any noise coupled into circuit from the USB power supply as well as give you more consistant 5v rail making the results between different people's equipment more comparable.

This would involve bringing back the power jack and connecting it to the Nano's Vin and ground pins. You just need to supply the jack with 6-12vdc there after. Given the simplicity of this mod, I'm planning on employing it on my V2 equipment once I get my parts together.

spenceraxani, 2018-06-29:
I’ve also read that the USB power can vary between that range but have never measured it to vary more than roughly 1% from a handful of different computers. I’m wondering if that range of voltages is an old figure. Earlier versions used the 5V regulator and required a DC power supply, however, we found that we always ended up simply using the USB ports anyways. Removing the DC barrels jack gave us room on the PCB to put in the coincidence connection, which was much more useful for making actual measurements. I guess it depends on what you want to measure — if you are interested more in photon counting, then this is one of a few upgrades that would help.

mmalluck, 2018-06-29:
I think the problem comes more into play with USB battery packs and wallworts than computer USB ports. The computer should have very uniform, clean, DC, as it's the same source provided to the other computer components. USB power packs and batteries, on the other hand, will have varying degrees of voltage and noise.  Checkout some of the comparisons between Apple USB plugs and the knock-offs. The difference in ripple current alone is surprising.

You could still use power from a USB source, but if you wanted to go that route, I would recommend using a boost/buck converter to ensure you get a nice clean isolated 5v source no matter what how you power it. If I were going to do that, I'd change out the Arduino to something like a [Pro Mini](https://www.sparkfun.com/products/11113) as the USB port wouldn't be needed at the Arduino and more or perhaps just start including the ATmega328 on the main board, no discrete Arduino needed.

There's no reason to use up valuable PCB real-estate for a power jack. As far as my simple mod is concerned, a panel jack and a couple of wires should be able to be shoehorned into the chassis without too much fuss, at least that's what I'm going to try.

I have a another question, why no decoupling capacitors near the op amps? In school it was drilled into my head that you should provide decoupling caps on the power lines of each of your ICs.