spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #12: Scintillator Size

lelipwn, 2018-07-05:
Hello Spencer,

In another issue post, you commented that the 1cm thinkness for the scintillator was chossen so that it would be easier to calibrate the detector to muons and not other background radiation. 

I happen to have however a 2.75cm * 2.75cm * 10cm plastic scintillator with both ends crystal clear (it's either a BC408 or BC412 not really sure). 

Shoud I just stick the SiPM in one end and wrap the whole thing to have a gamma + muon detector? Or should I cut a 2.75cm * 2.75cm *1 cm slab? 

Thank you making this awesome project,

L

spenceraxani, 2018-07-06:
Hi,

The electronics were developed for a 50x50x10mm^3 slab of scintillator, but it should work fine with the one you quote. You will see more light on average per event (since the SiPM covers a larger relative surface area), which means you will saturate the electronics more easily. If you are simply looking for counting purposes though, it should work fine. Keep in mind that your count rate will also be about 1/3rd that of our detector.

Thanks,

Spencer

> On Jul 5, 2018, at 5:20 PM, lelipwn <notifications@github.com> wrote:
> 
> M

lelipwn, 2018-07-06:
Hi,

I'm sorry, I don't understand. If I were to use the 27.5mm * 27.5mm *100mm why would the count rate be 1/3 considering the surface area is pratically double of the original one.

On the other hand if I used a 27.5mm * 27.5mm * 50mm the surface area is pratically the same as the original, would that be preferebly in order to not saturate the SiPM? 

I'm just worried it will saturate and not be at all accurate.

Thanks for everything,

L

spenceraxani, 2018-07-06:
Sorry, my answer was for cutting the scintillator to 27.5 x 27.5 x 10 mm^3. This would reduce the area by about a factor of 3. 

Your first issue might be mounting the PCB onto the scintillator, since the holes mounting holes are 30mm apart. If you use a thicker slab of scintillator, it also will not fit into the enclosure (which is fine). If you keep it at 100mm, then interactions at the edges might not trigger the detector — but that’s just speculation. 

My recommendation would probably be to use the 27.5 x 27.5 x 50 mm^3, to see what it looks like. I think it will probably be fine, you’ll probably end up with more light per event, but fewer events (since the downing muons will still only see 27.5 x 50 mm^2. This then gives you two pieces of scintillator if you plan on building another detector, and also allows you to cut it down to 27.5 x 27.5 x 10mm^3 if you chose. 

Spencer