spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #13: Solder preference

metapulp, 2018-07-11:
Hi, Spencer, do you use lead-free or tin-lead solder? Thanks!

spenceraxani, 2018-07-11:
Hi,

I personally like using 0.031”, 63% tin and 37% lead solder. 

Here’s a link to the solder that I purchase:

https://www.amazon.com/Kester-24-6337-0027-Solder-Alloy-Diameter/dp/B0149K4JTY/ref=sr_1_4?ie=UTF8&qid=1531342218&sr=8-4&keywords=solder+kester <https://www.amazon.com/Kester-24-6337-0027-Solder-Alloy-Diameter/dp/B0149K4JTY/ref=sr_1_4?ie=UTF8&qid=1531342218&sr=8-4&keywords=solder+kester>

Thanks,

Spencer

mmalluck, 2018-07-20:
I've been using a lead-free solder paste (95.5Sn/4.0Ag/0.5Cu) and a DIY reflow oven for the 5 boards I've put together. It works pretty well. 

A word of caution if using a reflow oven: 
Components like the SiPM may raise off the board slightly during the solder process and 'float' on the solder. While normally this isn't a big deal, the SiPM board only provides solder pads in two corners of the SiPM, so the SiPM may lift off the board ever so slightly an at an uneven tilt. This leaves the SiPM vulnerable to being damaged when the scintillator is installed.

I didn't notice that this happened and managed to crack one of my SiPMs (ouch!), but lesson learned. 

Either hand soldering the SiPM or having all 4 pads on the SiPM PCB would help avoid this.