spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #14: Polished vs unpolished scintillators

jgOhYeah, 2018-07-14:
Hello,
A quick question on the performance difference between polished and unpolished scintillators.

I have built two detectors. I was not quite game enough to begin with to heat polish the scintillators, so I left the surfaces as they were (one side glass like, the others slightly rougher and opaque, with the SiPM attached to the glass like side). Both registered around 1.5 counts per second.

Later, I had to take one of the sensors apart to fix a dry solder joint on the SiPM. While I was at it, I had a go at heat polishing the scintillator using a gas soldering iron with the tip removed, practicing on a bit of old acrylic first and being very careful not to burn the scintillator. This sensor now reads at around 0.6 counts per second. This seems to be the opposite as I expected.

The scintillators are made from Bicron 408 plastic according to the listing that I got them from and the instructions that came with them seemed to suggest that performance was better with them unpolished, although the instructions mainly seemed to focus on round scintillators attached to photo multiplier tubes, as opposed to square scintillators attached to solid state sensors with screws.

My question is: Is this difference in performance expected or have I made another dry solder joint or some other error somewhere?
Thankyou very much.

spenceraxani, 2018-07-15:
Hi,

The heat treatment was just to make the edges smooth and transparent to help with both the total internal reflection and the reflection from the aluminum foil. Typically with this versions of the detector, it’s not uncommon to see ranges varying from 0.8cps to 1.5cps depending on where you are located. This is due to the radioactive background of where you are located. 

If you made the measurement at the same location, I would be surprised with that large of a change. 1.5cps is typically on the higher side, and I wonder if there could have been a small light leak? 0.6cps is on the low side as well, but not unreasonable depending on how well the SiPM is coupled to the scintillator, the surface finish, and so on. 

I would have expected the count rate to be slightly higher with an improved surface finish. The best way to look for problems is to have a look at the test point connections. 

Hope this helps.

Thanks,

Spencer

jgOhYeah, 2018-07-17:
Thanks for the suggestions. I will look into them and the test points a bit more when I can get access to an oscilloscope.
The detectors were both in the same location at the same time when the measurements were taken (once in Melbourne and a few times in the Adelaide Hills on holidays). I have managed to get the detectors to work in coincidence mode and the slave unit registers a count rate of around 0.2 in the Adelaide hills. I might try re wrapping one of the sensors to see if that helps.

spenceraxani, 2018-07-17:
Hi,

0.2cps sounds very reasonable for the detector set in slave mode sitting next to the master.

Thanks,

Spencer

closed 2018-07-29 by jgOhYeah