spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #15: Test Connections

lelipwn, 2018-07-19:
Hello,

My detector is trigering at about a 0.126 cps which I know is pretty low. I am having problems determining the problem. Now I know that my scintillator isn't exactly to size with yours, being 3cm * 10cm * 1 cm so maybe that's why ( [Here](https://youtu.be/1XXfAtaSklA) is a short video of the 3 test connections. The initial change is me having trouble attaching the leads of the osciloscope, not the voltage jumping arround. While I'm seeing count rates of about 0.095 cps I am unable to see the peaks in the osciloscope. It's a cheap usb one, so maybe that's why. 

I did all the testing provided in the FAQ, my SiPM is reciving 29.5 Volts, and it does have the correct polarity. I am pretty sure it isn't random noise as it is pretty consistent, giving 0.095 cps in a period of 5 hours.

I don't know, if you think this numers are correct factoring the change in scintilator I'm fine with it, it just seems on the very low end.

`SIGNAL_THRESHOLD = 10`
        
`RESET_THRESHOLD = 15`

`LED_BRIGHTNESS = 255`

Thank's for all,

L

spenceraxani, 2018-07-19:
Hi L,

I would also start by raising the SIGNAL_THRESHOLD. A value of 10 is pretty low. Try bringing it up to 30-50 or so.  It could be that you are sitting below a small DC offset and the detector rarely resets (ie. the waveform has to dip below the SIGNAL_RESET value before it begins looking for another pulse). 

I think your next best bet would be to find an oscilloscope (the one you have might not be fast enough — probably something around 10+MHz would be fine, which might be close to your USB scope.) I have a hard time understanding what I’m looking at in the video you sent. Are you able to set the trigger on the USB scope? It looks like it’s just continuously running.

Since you have 29.5V, you need to make sure your SiPM is seeing something. You’ll have to check the signal on the BNC connection first. These are the raw SiPM pulses, and should be 10-100mV with a decay time of roughly 500ns. Once that is working, then moving onto the other test point connections will help diagnose the rest of the circuit.

Thanks,

Spencer

lelipwn, 2018-07-22:
Thanks for your help, 

I will wait till september when I have acces to a nice oscilloscope meanwhile, is it possible this isn't working right because I used plastic standoffs? Are they neccesary to complete the circuit between the SiPM PCB and the main PCB? 

Thanks again,

L

spenceraxani, 2018-07-22:
Hi,

Plastic standoffs will work fine. Two of the connections in the 2x3 header are for the ground. 

Thanks,

Spencer

lelipwn, 2018-07-22:
Hello,

I'm tring to find this information online but I'm not having any luck. How far does the SiPM "see" into the scintilator.  I found a nice article that compared angles of incidence, but nothing on how far a photon can triger "into" the scintilator and still be captured by the SiPM.

I re-wrapped and repolished my scintillator and this is the result. There has been an increase in the count rate, now is 0.218. I'm thinking all of this is just bad scintillator size, I will try to see if I can find the correct one. 

Thanks,

L