spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #16: 0.4 V from DC-DC booster circuit to SiPM

Pappy85, 2018-07-21:
I've built two boards with the same issue, there is 0.4 V coming out of the DC-DC booster for the SiPM instead of 29.5 V. I've checked each component in the booster circuit, including the connections on the LT-3461, and everything seems right. Has anyone seen this problem before? Any help diagnosing this problem is appreciated. Thanks!

Pappy85, 2018-07-21:
Of course - picture attached. Thanks for taking the time to look at it.

[image]

spenceraxani, 2018-07-21:
Hi,

It looks like the components are in the correct place, and the orientation is correct. But, I it does look like you might have some poor connections (for example: the top middle pin in the 6-pin header looks like it’s not soldered to the board, and maybe also the LED connection to the Arduino pin D3, maybe the top of R6). This makes me think that there could be a poor connection on the DC-DC booster as well. 

Could you start by melting and re-solidifying the solder on the booster legs? Just to make sure that each one of those connections is properly soldered to the PCB? 

It’s not obvious, but there could be a short between the two top right pins of the DC booster IC as well.

Since you aren’t seeing the correct voltage, the problem must be coming from within the DC-DC booster silkscreen outline.

Thanks,

Spencer

Pappy85, 2018-08-14:
Sorry for the delay responding, I was slowly working though various issues on the board. You were right that it was the solder joints on the DC-DC booster. I subsequently found some issues on the LT1807 legs as well. But after some persistence I got the system running and am counting muons. 

Thanks for the feedback here, as well as the youtube videos and instructions. I'm happily enjoying the detector. 

FYI - I don't know what others have found, but in my case, without access to physics department materials and a reason to buy in components in bulk, my completed system cost as least twice the quoted price of $100/detector. I only mention it as a fair warning to others considering building one. I think it was worth the cost though.

closed 2018-08-14 by Pappy85

spenceraxani, 2018-08-17:
Hi,

Glad to hear you got it working. 

The price is broken down in the Purchasing_List.exel spread sheet. We use this projects for teaching applications more so than for individual people. We build the detectors for sub-100$, but that’s typically in bulk for classes and outreach programs. If purchasing single components, the price does increase.  

Thanks,

Spencer