spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #20: SiPM/Scintillator failure then success

metapulp, 2018-09-03:
My detector was not functioning properly, and I decided the issue was likely the scintillator or SiPM or both. While unwrapping the scintillator, the SiPM became loose, and I was concerned an edge was crushed. Having the SiPM pressed up against the scintillator concerned me, so to cushion them I bought neoprene washers (90133A005) from McMaster Carr. 0.12" ID, 0.25" OD.

I had originally hand drilled the 4 holes before hand wet-sanding the scintillator using 600 to 2500 grit wet sandpaper and regular white ink-jet paper, but the second time around I touched the scintillator up with cotton flannel and the Novus system per these instructions: [https://eljentechnology.com/images/technical_library/Plastic_Polishing.pdf](url). This really contributed to the clarity. Spencer states elsewhere that all machining must be done before heat polishing. There are any number of materials that will fracture if drilled into at high-speed, so slow speed drilling while keeping the material cooled with water is what I typically do.

After a careful rewrap - wrapping the tinfoil like a miniature present and adding a second layer, I put optical gel on the SiPM and carefully screwed the SiPM PCB to the scintillator. I sighted the gap between the surface of the scintillator and the SiPM PCB to make sure I wasn't crushing the SiPM.

I rewrapped the scintillator with electrical tape.

I also added the neoprene washers to the short standoffs also.

This time everything works great. Photos added in case anyone else needs to use this approach.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

spenceraxani, 2018-09-07:
> I had originally hand drilled the 4 holes before hand wet-sanding the scintillator using 600 to 2500 grit wet sandpaper and regular white ink-jet paper, but the second time around I touched the scintillator up with cotton flannel and the Novus system per these instructions: https://eljentechnology.com/images/technical_library/Plastic_Polishing.pdf <x-msg://43/url>. This really contributed to the clarity. Spencer states elsewhere that all machining must be done before heat polishing. There are any number of materials that will fracture if drilled into at high-speed, so slow speed drilling while keeping the material cooled with water is what I typically do.

Hi,

Thanks for the photos. I’ve also found that using a 1/32” silicone rubber sheet works well for a spacer around the SiPM (see for example: 86045K67 from McMaster-Carr). 

Thanks,

Spencer

jimgallatin, 2018-12-04:
I used two layers of tinfoil and a layer of tape on the scintillator.  Since we are talking about 1/32” of cushion, would it make sense in your experiencefor me to simply loosen the four screws a bit to achieve that gap?

jimgallatin, 2018-12-04:
Also, as long as it is not crushed or damages is there any reason why the SiPM shouldn’t touch the scintillator through the gel?

spenceraxani, 2018-12-06:
Hi,

The face of the SiPM is fine touching the scintillator, but you don’t want to put too much pressure on it. Normally, I leave a small gap between the SiPM PCB and the scintillator, or I put a 1/32” silicon rubber sheet to give it the correct spacing. 

Hope this helps.

Thanks,

Spencer