spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #21: Problems with uploading to arduino and counting muons

rbaker314, 2018-09-05:
When the arduino is connected to the cosmic board I cannot connect the arduino to my computer to change the code. There is no USB recognized.  When I disconnect the Nano from the board, it connects fine. Is this normal? I have uploaded the software with the Nano off the board and then added it to the board but I get no counts. All of the voltage tests are good.

When the power is connected to the arduino the oled shows the Cosmic Watch logo and the clock advances every second but the count does not change after counting once. I changed the const_threshold from 20 to 15 but no change.

Not sure what to do.

Thank you,
Robert Baker

rbaker314, 2018-09-06:
You mention in the Youtube video that two of the Arduino Nano pins do not need to be connected. Do those pins need to be cut?

spenceraxani, 2018-09-07:
Hi,

There are several pins on the Arduino that aren’t used at all, and they can just be left unconnected if you like. Connecting them doesn’t hurt though.

Thanks,

Spencer

jgOhYeah, 2018-09-08:
On one of the detectors I built, I had a similar problem where the arduino connected to the computer off the muon detector board but only connected sometimes after I had soldered it to the board. I used some cheap clone arduinos and I think that maybe there was a dry joint on or around the data lines in the builtin usb connector or the usb serial converter chip on one of them. I retouched the joints around the connector and it can connect to the computer now.

spenceraxani, 2018-09-08:
Hi,

I’ve had a few instances where I’ve had to unplug and plug back in the Arduino, but that’s about it. 
If you are using a Mac, you need to have the CH340g driver installed. 
If you are using a Linux OS, you have to change the permissions on the port to allow it to communicate. 
If the Arduino IDE says that the port is in use, that means that you are already talking with it through another program (such as our import_data.py). 
Otherwise, perhaps there is an issue with the Arduino — although I don’t think I’ve ran into this specific problem before. I like the idea of touching up the soldering joints on the USB mini port just in case.

Thanks,

Spencer

rbaker314, 2018-09-20:
Hi,

I cut off pins connecting pins 1 and 0 of the arduino.

Still have problems connecting to arduino with computer when plugged into Micro Watch board.

Below is the error code from the arduino IDE when I try to upload the code.

Arduino: 1.8.7 (Windows 10), Board: "Arduino Nano, ATmega328P"

Sketch uses 20670 bytes (67%) of program storage space. Maximum is 30720 bytes.
Global variables use 1138 bytes (55%) of dynamic memory, leaving 910 bytes for local variables. Maximum is 2048 bytes.
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega328p -carduino -PCOM3 -b115200 -D -Uflash:w:C:\Users\jwise\AppData\Local\Temp\arduino_build_565452/Arduino_Code.ino.hex:i 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x20

avrdude done.  Thank you.

An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Thanks,

Bob

closed 2018-09-20 by rbaker314