spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #22: Issue with consistent read out & issue with verifying parts

80years80, 2018-09-19:
Hello, 

Could you please add a parts legend? E.g which resistor values on the board go with which value.

I am having issues having the single detector reading out data consistently. The Oscope data for the most part seemed good, but TP2 did seem noisy when I lowered the trigger threshold. The issue I am having is nowhere lists the resistor values for it. The picture of the button shows a 300 ohm resistor but we bought a 249 ohm resistor for R3 as the parts list suggested. I did try retouching the legs between the peak and amplitude detector. 

It seems after running it several times on average it seems that the count increments every 6.66 seconds after letting it run for several minutes.  However it is by no means consistent sometimes it will go minutes without blinking then it will rapidly start blinking/counting and catch up. 

Here is my recorded Oscope data, when I looked at TP2 for the second time it seemed that once the peak was detected it would go crazy up and down, but when it was slightly higher it seemed normal.

spenceraxani, 2018-09-20:
Hi Emre,

I think the file you are looking for is called the SMT_reference.pdf located here:
https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/tree/master/PCB_Files <https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/tree/master/PCB_Files>

Did you lower the Arduino threshold? Setting it around 50ADC counts should be fine. If you zoom into the baseline of the TP2 it will look noisy, but the pulse should look smooth. The resistor value should be 249, can you point me to the picture that shows it as 300Ohms? 

You probably still have an issue with the circuit. The count rate should be roughly 1Hz, and the pulses should arrive randomly. Your TP2 should look similar to TP1 but amplified by a factor of ~20. TP3 stretches the pulse in time by roughly a factor of 1000x, so you’ll have to zoom out. If any of these still look odd, you’ll have to keep trouble shooting your circuit.

Thanks,

Spencer

80years80, 2018-09-20:
Hi Spencer,

Thanks for directing me to the parts link I am not sure why I didn't check under pcb files, my bad. 

I did try playing around with the threshold some and it did not really seem to improve the situation. 
Here is the link to the image: 
https://raw.githubusercontent.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/master/Pictures/Bottom%20view.jpg

I agree with you that at this point it still seems to be a circuit error. The pulses seem to arrive randomly so I think at least it is on the main pcb. 

I think a sample of the oscope data I was getting and observing seemed like you described it. I think at this point from the observed behavior that either way it is within the circuit, hopefully it will be quick to verify the correct parts are in place. 

The measured random sample data I was getting was: 
TP1:  30mV, 80mV, 100mV, 40mV. Settings were 20 mV, 1 uS
TP2: 2V, 1.6V, 2.8V, 1.2V, 0.5V Settings were 1V, 2.50 uS
TP3: 1V, 1.5V, 0.8V, 0.4V Settings were 1V, 2.50 uS

Thanks, 

Emre

80years80, 2018-09-21:
So I went through the circuit again and noticed a few things: 

In R5 and R20 in the circuit supposed to be 10k components both measured about 5.05k, from looking at the schematic I don't really see any reason why the resistances should be reduced in the circuit as if they are in parrell unless maybe a cap got busted?. R6 and R7 both read 50k. but I think that might be normal for that part of the circuit

Another thing I noticed is the 10uF cap C4 on the main PCB when I measured the same component used for it with a volt meter it returned 8uF, I am aware the capacitor has a tolerance of about +/- 20% which could be the cause of this. Is this possibly due to the wrong component being bought? Or most likely the tolerance is the reason?

spenceraxani, 2018-09-21:
Hi,

If you measure the resistance across R20 while the board is populated, you’ll measure a reduced resistance since you are also measurement the R3/R5 components in parallel. ie. 1/(1/(10000+249)+1/(10000)) = 5.06k. The R20 resistor isn’t necessarily important, it was more an artifact from optimizing the circuit. The reason you measure a lower resistance across R6 and R7 is the same argument; you also have to take into account the resistor R24.

C4 and C5 are just capacitors used to reduce noise on the VCC (power line). The values of these capacitors aren’t very important though. Any uF-scale capacitor will have a similar effect. I would bet that the reason that you read 8uF is due to the tolerance that you point to.

Hope this answers your questions.

Thanks,

Spencer

80years80, 2018-12-26:
Hello, after coming back to this project since I have time again I started trying to note data again and the OScope Data seems correct to me and should work fine with the analog pin it is connected to. 

When I did open the serial monitor I did notice something weird, that all the counts it got are around 260, and that the sipm voltage is always about 44mV. It seems the detector only counts when this occurs. It seemed like from your sample data in the instructions that this should vary much more than my data is. 

Do you have any idea what in the circuit could be causing this behavior? 
Here is the data I am talking about. This was with the signal threshold set to 50:
##########################################################################################
### CosmicWatch: The Desktop Muon Detector
### Questions? saxani@mit.edu
### Comp_date Comp_time Event Ardn_time[ms] ADC[0-1023] SiPM[mV] Deadtime[ms] Temp[C] Name
##########################################################################################
Adler
1 41 231 39.55 81 14.13
2 7925 259 44.69 8229 14.13
3 9310 255 43.96 9691 14.13
4 12477 255 43.96 12971 14.67
5 14054 255 43.96 14623 14.13
6 16482 255 43.96 17126 14.67
7 37353 268 46.26 38795 14.13
8 39439 259 44.69 40955 14.13
9 39697 248 42.69 41213 14.13
10 41383 263 45.37 42974 14.13
11 43493 255 43.96 45162 14.13
12 46138 255 43.96 47922 14.67
13 51703 255 43.96 53678 14.13
14 54288 248 42.69 56379 14.35
15 61078 259 44.69 63437 14.13

Thanks,

Emre

spenceraxani, 2018-12-27:
> Questions? saxani@mit.edu <mailto:saxani@mit.edu>

Hi Emre,

That’s pretty interesting. 255 is what you would get if you were maxing out an 8-bit ADC (2^8 = 256, or 0-255). I see however that there is values larger than 255 though, so it could just be a coincidence. This is a new one for me.

The count rate is also pretty low, and the dead time is way too high. It’s likely that there is a problem with the circuit. Did you test all three test point connections using the oscilloscope? To me, it looks like there is some DC offset of approximately 44mV.  The detector looks like it's getting stuck after triggering, then waiting for the pulse to return below the 50 ADC count trigger. However, since there might be this offset, it never (or rarely) sees the ADC count drop below 50. This would cause you to almost always get the same reading, and inflate the dead time. 

I can’t identify what could cause this without checking the waveforms on the test point connections. Can you describe what you saw when hooking up to the oscilloscope?

Thanks,

Spencer

80years80, 2018-12-28:
Here's the videos:
BNC: https://youtu.be/cT_H8t7dOYk
TP1: https://youtu.be/fEiPCxk4SDs
TP2: https://youtu.be/b9EeosZQ5G8
TP3: https://youtu.be/dR684vRVL2c

It seems that there is as you suggested a DC offset of about 50mV or there is some sort of oscillation going on.

spenceraxani, 2018-12-29:
Hi Emre,

It looks like you are close. I can see large fluctuations in the baseline that are likely causing the problem. The good news is that the rest of the circuit is likely correct. 

While it’s extremely noticeable in TP2 and TP3, even the BNC connection looks rather noisy to me. I haven’t seen this one before, but my best guess would be check a few things.

1. Make sure it’s light tight. Maybe wrap in a black cloth or bag to check.
2. Replace the op amp. This is my best guess to the problem. LT will send you a free sample if you have an account with them.
3. Check the power to the SiPM. A multimeter might indicate that it’s 29.5V, but maybe there is underlying noise that you cannot see. You can look at the DC-DC booster waveform by AC coupling it to the scope. I see that on your screen, top right, it says that you are DC coupled. 
4. Sadly, it could be an issue with the SiPM. But I this this is unlikely.

You can also zoom out in the x-axis, to see if the problem is cyclical/. If it’s 60Hz, it might be from the lights. If it was 3MHz, it would be the booster, but we would see that in the current video.

Hope this helps.

Thanks,

Spencer