spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #24: SiPM purchase source

pedelen, 2018-10-29:
Sensl was taken over by ON Semiconductors. Their website for purchasing a SiPM is no longer functional. I have so far not been able to reach ON Semiconductor to see if they well support selling small numbers to individuals. Does anyone have an extra SiPM that they might be willing to sell?

pedelen, 2018-11-02:
I was able to reach a contact at ON Semiconductor about this. I cc'd you on my email to him. Hopefully this will resolve the issue.

 Best regards, Paul

spenceraxani, 2018-11-02:
Thanks for looking into this.

Spencer

pedelen, 2018-11-14:
I am working with Digikey on getting the C-series SiPMs  as a non catalog request. If that works out, I will post again with the procedure and cost. The part number will be MICROFC-60035-SMT-TR1

msrooks, 2018-11-24:
Mouser has some on order: Mouser#: 863-MFC60035SMTTR1 Mfr#: MICROFC-60035-SMT-TR1 $72.92

pedelen, 2018-11-25:
Thank you that is good to hear. I had originally tried Mouser but they said six month lead time. It looks like Mouser will be about $10 cheaper than Digikey. Will let you know when I get final invoice.

msrooks, 2018-11-25:
No Problem! They originally said it would ship mid December but it is in transit now.

pedelen, 2018-11-25:
Excellent. Mouser is about $10 cheaper than Digikey quote of $82.42. If we all go with Mouser we should be able to negotiate them down a little and be closer to the original cost of $60 through Sensl

pedelen, 2018-11-25:
Will close this issue as it appears source noted above by msrooks is available and priced close to the original price deal from Sensl. As per msrooks:
Mouser has some on order: Mouser#: 863-MFC60035SMTTR1 Mfr#: MICROFC-60035-SMT-TR1 $72.92

closed 2018-11-25 by pedelen