spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #25: sketch failing due to errors

northern-nuke, 2018-10-31:
Using Windows 10

error message:
[arduino_fail.txt](https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/files/2535908/arduino_fail.txt)

Also having difficulty sourcing LT3461A -- trying direct from China.

thanks
Bryan

per1234, 2018-11-02:
You're trying to compile some HTML code used to display an Arduino sketch in the browser. You've utterly failed at copy/paste.

spenceraxani, 2018-11-02:
Hi,

Could you elaborate on the last comment. What piece of code are you trying to run?

Thanks,

Spencer

northern-nuke, 2018-11-03:
still trying to get cosmic watch code into arduino.

I am now pasting naming file in sketch as run once.
I get an error re syntax.

sketch_nov03d:19:37: error: a function-definition is not allowed here before '{' token

 boolean eeprom_is_addr_ok(int addr) {

                                     ^

sketch_nov03d:23:76: error: a function-definition is not allowed here before '{' token

 boolean eeprom_write_bytes(int startAddr, const byte* array, int numBytes) {

                                                                            ^

sketch_nov03d:395:1: error: expected '}' at end of input

 }

 ^

exit status 1
a function-definition is not allowed here before '{' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

I'll get there eventually. Still waiting on parts deliveries.

per1234, 2018-11-04:
The error message's line numbers still indicate that you didn't copy the sketch correctly. The best thing to do is to just download the full repository:

1. Go to the repository homepage: https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2
2. Click "Clone or download".
3. Click "Download ZIP".
4. Unzip the downloaded file.
5. (In the Arduino IDE) File > Open > select CosmicWatch-Desktop-Muon-Detector-v2-master/Arduino/Naming/Naming.ino from the unzipped folder > Open

If you do want to only grab a single sketch from a repository then you need to first click the "Raw" button on that sketch page. After that you can either use your browser's File > Save menu option or else just copy and paste the text from the browser. Be aware that some Arduino sketches are made of multiple files (though not the Naming sketch) so if you only grab a single file of the sketch it won't work.

northern-nuke, 2018-11-14:
I've returned from other distractions.

Previously I was trying to get the code into an already open Arduino IDE window. I changed to opening the naming.ino file as you suggested, including the library files, and it compiled!
Now I have the next problem -- the IDE detects the Arduino and identifies it. Alas the IDE tries to upload and fails to "sync" giving up after ~13 attempts. I have installed the Win CH340G driver. I am using an old Dell 32 bit laptop running W10. (I previously had FTDI chipset serial port drivers installed.)

My LT3461A chips arrived from A-Grade Electronics at an embarrassing price. I'm still waiting on the Elecro front and rear plates for the enclosure.

northern-nuke, 2018-11-15:
Thanks for your encouragement ... the "old bootloader" option was what it took. Software is in. 29.5 V ... now I have to finish it and see if it counts!

northern-nuke, 2018-11-17:
I cut the mis-matched Vcc and Gnd pins on the OLED board connector and hard-wired the Vcc and Gnd. Amazingly enough it survived and displays. I get counts (and LED flashes) only when I tap the scintillator. 

I have some more work to do yet ... still waiting on the Elecro front and rear panels.

northern-nuke, 2018-11-21:
Have 1 of a pair of CosmicWatch V2 running. I find that this unit generates a single count on startup and counts for ~ 2 h and reaches a total of ~ 472 or ~ 380 and then stops counting. The display and timer functions continue but no more counts for hours. 

Removing power / reconnecting or pressing reset button causes a restart and one count, but no more counts. Removing power and then trying again the next day provides a similar sequence of events. I have the boards in the case without the end-plates (somewhere in the mail now exacerbated by labor problems at Canada Post) secured with rubber bands.

The other unit starts with a flash of the green LED on the Arduino -- but no count. It did something similar earlier reaching ~56 counts before quitting. Alas in trying to resolve the SIPM connection ONE MORE TIME I succeeded in removing both solder pads from the board.  Talk about skill! 
I drilled the through-holes out and routed a wire through each soldered to the SIPM and the respective resistor. I have yet to get this unit counting again. It does not start with a single count unlike its sibling.

I have yet to impose on my friend with a 'scope.

northern-nuke, 2018-11-22:
I tried to upload the OLED code again in the hopes of updating the library in one unit. This failed with an error and now the code appears to be running but the OLED display is not.
How do I perform a fresh re-load of the code into the Arduino?

northern-nuke, 2018-11-22:
I should have mentioned that I moved the non-repaired SIPM card to the board that was not counting for a test. Following my code mangling above, it appeared to be counting (green and white LEDs flashed). The "other" unit is not counting. This is consistent with suggesting that my "repaired" SIPM board is not functional.

northern-nuke, 2018-11-26:
Hope you had a great Thanksgiving holiday.
Is the 29.5 V or 29.9 V too high for the SIPM? When I plugged the "coincidence" cable in to power both  units, the unit that counts seemed to work again for a while. I get the same voltage levels with either a PC USB port or a portable USB battery. My 4.6 V is 4.7 V. 
Somehow I managed to pull the connector off the micro SD card -- haven't a clue what I did, but it pulled the pin out of the connector and pulled the pad off the circuit board. I need a "solder sucker" to remove the board to complete the "repair".

spenceraxani, 2018-11-26:
I’m actually at the South Pole right now, but we did have a great Thanksgiving diner a few nights ago.

I’ve operated the SiPMs at 29.9V before, however there is a spread in breakdown voltage of the SiPMs. I’ve never ran into problems running at 29.5V.

I would also recommend using solder wick rather than a solder sucker. The solder suckers can be a bit forceful when working with surface mount components and can sometimes rip the pad off with the solder.

northern-nuke, 2018-11-28:
OLED displays now working on both units (a patch wire that I thought was soldered came free - fixed that.)
noticed that when the 2 units are connected via the "coincidence cable" neither counts. "Unit 2" counts sometimes when it is powered alone. Perhaps Unit 1 has an oscillation problem?
Software reload was straightforward once I checked the COM port and made the change.
I've been using silicone grease as an optical coupling agent. Moen sells a small container with this for O-ring lubrication when maintaining plumbing. I tested it with pieces of PMMA and observed that it appeared to provide good optical performance. It's easy to find and inexpensive compared to liquid optical coupling fluid.
More tests to do.

closed 2018-12-10 by northern-nuke