spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #26: Face plate screws

msrooks, 2018-11-25:
Greetings, 
I don't see a listing for the face plate screws. Looking for a recommendation please. 

Regards, 
Michael

msrooks, 2018-11-28:
Perfect thank you! My case did not come with any screws. I'm also waiting for face plates to arrive.

northern-nuke, 2018-12-08:
I thought they were #6, but they are smaller. The hole in the case extrusion is a bit larger than the drill for tapping 4-40. I tapped the holes. This made installing the (non-4-40) screws easier and generated less we bits than the #6 "sheet metal" screws would have. They are a bit homely, but they went in.

msrooks, 2018-12-08:
I also noticed this after buying the #6 screws, but thankfully I also found the screws that came with the case hidden inside the box. Both cases did indeed come with screws.

msrooks, 2018-12-08:
Probably okay to close this issue @spenceraxani my mistake for opening it. The Enclosures & Cases folks provided the screws.

closed 2018-12-08 by msrooks