spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #27: Scintillator Malfunction

jimgallatin, 2018-12-03:
Spencer, I am switching from your email to this account to declutter your life.  

I have built two sets of SiPM and Main PCBs, and think I have them working.  The voltages are correct, and there is current throughout.  (I only have 29.3V on one PCB’s SiPM connectionbut I don’t see that as a problem.)  Soldering in good shape after some redos and replacements.  However, I am not getting a signal through the BNC, and no counts on the screen or through the Python data reads.

Since my first two scintillators were less than perfect, I am assuming that my core issue is the Scintillator or Scintillator-SiPM connection.  I plan to assemble a third combo but want to get some adult supervision before my third try.  (I used to work with NVG technology but this is way different.) 

Please excuse any and all ignorance here.

1.  I am confident I have wrapped the Scintillator well in one layer each of tinfoil and tape.  Should I move to two layers of each as some have suggested in these comments?

2.  My Scintillators are clear but were very slightly warped during heat treatment. Think concacity.  Issue?

3.  It is conceivable that the SiPM is not super flush against the Scintillator, but I did use a good amount of optical gel as you recommended.  How flush does it need to be?  I have been assuming the muons have nowhere else to go with all the wrapping.
.

jimgallatin, 2018-12-03:
Thanks again for such a great project.  Jim Gallatin

jimgallatin, 2018-12-03:
And meant photons not muons.  Sigh.

jimgallatin, 2018-12-04:
Spencer, I made another scintillator and fixed the issues of concern.  Onwards.

spenceraxani, 2018-12-06:
Hi Jim,

The scintillator is normally not an issue, as long as the face that interfaces the SiPM is clear. Having a concavity won’t hurt either, the optical gel should take care of that. I typically wrap the scintillator with a single layer of tinfoil, ensuring that it doesn’t short the connections on the SiPM, and then wrap the full piece in 2-3 complete layers of black electrical tape. Light-leaks are likely on the edges. I typically wrap it similar to a present. 

It’s possible that the SiPM isn’t flush against the scintillator. This will reduce the number of photons that you observe on every interaction. I tighten the SiPM PCB such that there is a gap roughly the size of the SiPM, or I put a 1/32” spacer around the SiPM and tighten all the way. If you are concerned with the tightening, you can always tighten the SIPM PCB to the scintillator without the tin foil to check how close to go, then where you are comfortable, wrap it in tin foil + black electrical tape. 

Next time you open it up, do you want to take a picture of the PCB and the SiPM so that I can check that everything looks good?

Thanks,

Spencer

jimgallatin, 2018-12-06:
Hey Spencer,

I am taking a break for exams but will be back in the Detector business next week.

Reading your comments, I am pretty sure I got the Scintillator and SiPM right.  I made a third Scintillator with great care and double wrapped the tinfoil and tape.  It is very clear and I used a good amount of gel as in the YouTube video.

Using either of my two “best” Scintillator and SiPM/PCB combinations, when I connect the Detector to a scope through the BNC, I am now getting identical triggering events, with scope settings at the designated amplitudes and rates, and identical accompanying waveforms.  However, I can’t make the waveforms match the ones on the website.  I am using a standard Tektronix 2000C scope in video mode, and vertical/horizontal settings at 20mA and 500ns, but am not experienced in using oscilloscopes.

When I get back to work, I will download the readings into Python to see what the data says.  I will send you pictures of everything after I check the Python data.  

The Arduino and Python codes are running fine, and I have the correct voltages and current throughout the Main PCB (I will reconfirm that again before sending pictures).  I have checked and, where not robust, resoldered every component.  I have the OLED working fine but have only gotten a handful of counts.

As you recommend, I will doublecheck check the resistance across the SiPM when I undo everything to send the pictures.

FYI, I found that it was best to heat treat a precut scintillator before gently drilling the four holes after the scintillator cools.  It is easy to heat the plastic so much that the predrilled holes close up.

Thanks again,

Jim

closed 2019-01-06 by jimgallatin