spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #29: Getting counts only when soldering iron comes into contact with capacitors connecting thenpeak detector to the Arduino.

jimgallatin, 2018-12-13:
Spencer,

I have a good looking pair of PCBs, SiPM and Main.  I have the appropriate current throughout.  But no counts.  I went through the Main PCB components again touching up the solder connections and made the weird discovery that when I touch the soldering iron to the capacitors between the peak detector circuit and the Arduino, or a capacitor in the peak detector circuit, the OLED shows a batch of counts.

I plan to replace the components where this happens under the theory that when the soldering iron comes into contact with the circuit that it closes an otherwise open circuit.  Does that make sense?

Once I have gotten through Christmas I will build a third, and hopefully perfect set of PCBs and send them to you for review.

Happy Holidays to you and the team in Antarctica,

Jim Gallatin

spenceraxani, 2018-12-13:
Hi Jim,

It’s likely that you are just inducing a current into the circuit which is causing noise that triggers the detector. Soldering irons generate heat, which means they are passing a lot of current which can induce a voltage in a wire. I doubt that this is what is causing the problem.

When you saw ‘appropriate current,’ do you mean that the DC booster measures 29.6V?

Thanks,

Spencer

jimgallatin, 2018-12-13:
Thanks.  Makes sense since the light didn’t flash ....  OK, onwards then.

On this Main PCB (#1) I can trace 4.7V through the non-SiPM circuit components, and have 29.5V across the GND and HV pins on the SiPM connection.  And the Arduino code loaded fine and the OLED does its thing when plugged in.  That seems to me to say that I have a functioning Main PCB.

After Christmas I will take the Detector apart and send you pictures of the boards.  I will also redo the scintillator-SiPM interface as carefully as I can.  And I will test the current across the SiPM connection, which I haven’t done so far.  I will also build one final Main PCB.

I recall from my work with night vision (using glass boules and PMT tubes) that all sorts of tiny imperfections in the manufacturing process, temperamental machines, or slight intolerances in assembly would render the final NVGs unworkable, and every aspect of this project is new to me, so I am a long way from giving up.  Just more than a little over my allotted four hours.

 Jim

spenceraxani, 2018-12-13:
Since it sounds like everything is in place, you should be able to narrow down to +3 components which could be causing the problem using the test points. If any of them look odd, it tells you that that is where the problem is. 

I think I go through testing the test points here:
https://www.youtube.com/watch?v=Ck24HGrjBfY

jimgallatin, 2018-12-13:
Thanks.  Will do.

closed 2019-01-06 by jimgallatin