spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #31: Super simple form, fit, fyctuion

jimgallatin, 2018-12-26:
I’m sure this is a rock dumb question but does it matter functionally that I installed the Arduino

jimgallatin, 2018-12-26:
With the long and short legs put on the wrong side of the main PCB?  I have Arduino on the correct side and can install everything else still so I don’t have a fit or form issue.  If so, I now have a great looking set of boards and functioning scintillator.  Once I receive a component I ran out of I will send pictures of the boards, connect it all, and report results.

I have successfully gotten appropriate counts but after an hour something stopped the, so I built a third (and final) Main PCB.  Almost there.

Happy holidays,

Jim

spenceraxani, 2018-12-26:
Hi Jim,

Sorry, I’m not sure I fully understand. Do you mean that the header to mount the Arduino to the PCB is just upside down? If that’s the case, it doesn’t matter. Also, did something happen after running for an hour? If everything is built correctly, they should run for ~40 days without a problem. After 40 days the variable holding the timing information will roll over. 

Thanks,

Spencer

jimgallatin, 2018-12-26:
Hey Spencer,

Yup, I installed the header upside down (talk about feeling foolish).  Figured not a problem but good to confirm.

After an hour it just stopped adding counts.  I am betting that something shorted or lost the connection when it was an unstable piece of soldering or whatever.  I was glad to get that far.

I should get my new components (Line Item 13 in the list) in a few days and hope to finally finish this phase of the project.

Jim

msrooks, 2018-12-31:
Hey Jim, 
I had a detector that would start and stop sometimes when I put pressure on the scintillating block. I finally figured out that it was just a light leak in one of the corners. I wrapped another piece of black electrical tape around the corners. This resolved the problem. 
Best, 
Michael

jimgallatin, 2019-01-02:
Mike,

Thanks for this.  I’ve been away.  It makes sense.  I’ll retape my scintillator.

Jim

jimgallatin, 2019-01-04:
Thank you Mike and Spencer.  I have a solid signal through the BNC connection that looks exactly right, 20mv and 500ns, less than 10 MHz (I’m guessing .5-1 MHz).  Awesome.  But I have work to do.  I have no signal from TP1 (I am assuming a bad solder joint), a nice signal from TP2 (mimics the BNC signal when running together), and no signal from TP3.  But I am super happy to get this far.  Happy New Year.  Jim

closed 2019-01-06 by jimgallatin