spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #32: LT 1807 Op Amp heating up

jimgallatin, 2018-12-27:
Spencer,

I have finished a new Main PCB (I’ll send pictures next).  I am pretty confident that the soldering is adequate.  I have a solid scintillator and SiPM combination.  I have 29.4V at the SiPM HV/GND points.  I thought I was fresh out of reasons for why the resulting detector won’t work when assembled and plugged in.

And yet.  When I assembled and plugged in, everything lit up, screen came up, etc.  However, within seconds after plugging it in, the LT 1807 became too hot to touch and faint smoke was emitted.

To be sure it wasn’t that particular 1807, I replaced it.  Same result (didn’t wait for smoke).  I assume I have a short!?  Any ideas?

spenceraxani, 2018-12-27:
Hi Jim,

Was it orientated in the proper direction (pin 1 should be orientated such that it is on the pad with the circle)? A short across the legs of the opamp could also cause this, but since you’ve seen it twice, it’s unlikely. A short (or 0Ohm resistor) across R3,R6 or R20 will also cause this.

Thanks,

Spencer

jimgallatin, 2018-12-27:
Thanks.  To be clear, are you saying that 1) a short across (say caused by too much solder) or 2) erroneously using a Zero Ohm resistor for R3, R6, or R20 could cause this result?  I will go back and double check the resistance of each of those components.

On the orientation, I am pretty sure I have it right as I have oriented it same way on the prior boards.  There is a little squiggly thing in one corner and I am treating that as the number one leg corner and placing it by the circle.

rogerio-acioli, 2019-01-04:
Hi, Jim!

I had the same problem at my first doing the main PCB. What I found was that I probably made a short between the legs of the LT1807. It really got too hot and emitted smoke. What I did was rebuild every piece by making sure it component was right and trying to make a really good job at the soldering of the opamp!

I hope you can make it work!
Thanks, 
Rogério

closed 2019-01-06 by jimgallatin