spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #34: Booster voltage first too high, now too low

theostoelzl, 2019-01-05:
Hi Spencer,

I'm having a little trouble with my Main PCB. I fully populated the PCB and plugged the Arduino into my power bank. (I tried with a phone charger as well, same result.) At first, I measured 4.6 V between the HV and GND pins, after which I resoldered the legs of the booster, which gave me about 33 V output. So I've resoldered a few legs again and now I'm stuck with something between 23.5 and 24.5 V, varying when I put some new solder on one of the legs - or take some away - and apparently also when room temperature changes (I just closed my window and voltage dropped from 24.5 to 23.5 V). Could I power the SiPM with that voltage, and do you have an idea where my trouble might be originating from?

Plus, I noticed that the inductor gets really hot while plugged in, should I worry about that? The booster heats up as well but not as intensely as the L1 inductor.

Thanks a thousand,
Theo

spenceraxani, 2019-01-05:
Hi Theo,

Nothing should even get noticeably warm to the touch, so you probably have a short somewhere drawing a lot of current. My first bet would be that there was a short between the booster legs (but it sounds like you’ve already tried this). My next guess would be a short or misplaced component somewhere in the booster circuit. The ratio of the resistors R1 and R2 determines the output voltage, so I would start there. As a firs step, I would take your soldering iron and go over each solder joint again, melting and resolidifying the solder to make sure you have a good connection. 

I would also be careful when plugging it in. The Arduino is the one supplying the power that is being thrown away to heat. It’s only able to supply a few watts, and you really don’t want to overload it (it’s just a big pain to replace).

You can also send me a photo of your board and maybe I can spot the issue.

Thanks,

Spencer

theostoelzl, 2019-01-05:
Thanks for your prompt reply!

I've already added the photo above.

I tried melting and resoldering, nothing changed; I did, however, remove some solder from the L1 inductor and my voltage went up to ca 25.8 V. The overheating remains, though. I also measured the resistance of R1 and R2, for R1 it says 10 kOhm and for R2 214 kOhm. Could the 12 kOhm difference in resistance be the reason for the missing voltage, so should I just try another 226k resistor instead or could the difference result from something else than a faulty resistor?

Thanks
Theo

spenceraxani, 2019-01-05:
The L1 component should not get hot, so I think we’ll have to figure out what’s causing this before trying to get the voltage to the correct value. The formula for the output voltage is 1.255(1+R2/R1), so if the resistance of R2 is 214, you should still expect to see 28.1V. Also, it’s pretty tough to determine the resistance of a resistor that is mounted on a circuit board since there are often other paths that the current through the board. For example, if there was a 4k resistor somewhere downstream of R2, that went to ground you would measure the resistance to be 214 rather than 226. This resistance could even be internal to the LT3461A. 

I doubt this is the issue, but I’m unable to tell in the picture. Is the booster mounted in the correct orientation? I don’t think you would see any voltage if it wasn’t, but I want to check that it was originally mounted correctly. 

I can’t identify an issue with any of the soldering from the picture, so if you’ve already redone all the connections there, we’ll assume it’s good. I also can’t tell the values of the capacitors since they are unlabelled, so there could be a chance that one of the capacitors is the wrong value. The only capacitor there that could cause issues is C1 — the others are only for reducing noise and you can ignore them. There’s also a chance that the booster is broken and causing problems (I’ve had roughly a 2% failure rate on the boosters). Do you have an extra one to swap in to test?

I see from your images that you also only connected the pins on the Arduino that had the dot beside them — I would connect them all just in case I missed on (this isn’t causing your problem, but might save you some troubleshooting later).

For the future, I would also place your resistors with the dark side up, so that you can read the values of all the resistors on the board. 

Spencer

theostoelzl, 2019-01-05:
Thanks for the tips, will do!

My bad, it really isn't visible at all in the photo, I believe I soldered it in the right orientation, the little arrow is pointing to the dot on the board.

I'll check the capacitor. But unfortunately I haven't got another booster so for now I'll have to trust that it works fine.

Theo

theostoelzl, 2019-01-06:
Hi, so I replaced C1 with the redundant 226k capacitor from the order and so far nothing has changed. I usually measure a voltage between 25.5 and 26.5 V now. However, I noticed that the voltage in the inductor between one of the soldered sides and the windings amounts to about 180 V, I just have no idea why it's _that_ high - or is that usual behaviour (considering the overheating I suppose no, but I'm asking as a complete electronics-anti-expert)? Could I have soldered the inductor wrong or might it be corrupt by default? I simply cannot find a short anywhere (else).

Theo

theostoelzl, 2019-01-06:
Sorry, I meant C1 with a 22 pF capacitor of course.

I'll definitely try that. I'm afraid I can't see your attached photo, though.

Thanks a lot, especially for taking time even on a Sunday,
Theo

spenceraxani, 2019-01-06:
Hi,

Sorry, I guess images don't pass through to github. 

[image]

spenceraxani, 2019-01-06:
> Now I've dismounted C2 and L2, plugged the device in and the inductor still heats up immediately. Between the pad and GND I measure either 14 V or 37 V, depending on how I connect the electrodes.
>
> Theo

Interesting. Next step will be to replace the LT3461A.

theostoelzl, 2019-01-06:
Thanks a bunch, already ordered a new one. Hoping for the best.

Theo

theostoelzl, 2019-01-11:
Hi again,

So I replaced the booster and the same issues persist.

Thanks,
Theo

theostoelzl, 2019-01-11:
Unfortunately, I have already replaced L2 with the redundant, unused one (so that _should_ be working fine).

Checked, voltage amounts to exactly 4.6V between 5V and GND.

Theo

spenceraxani, 2019-01-11:
Could you also solder all the connections on the Arduino pins and make sure non of them are bridged? After you are adding solder to the pins, can you send me an updated photo of both sides of the PCB?

The only other components in that part of the circuit are the two resistors and the single capacitor. It’s unlikely that there is a problem with those as long as they are the correct value. Incorrect values will change the output voltage, but you wouldn’t see any fluctuations — just the wrong voltage reading. I also know that you replaced the 22pF cap and touched up all the solder joints.

It could be that there is some other part of the circuit drawing a lot of current, and not supplying enough current to the DC Booster circuit. This could cause fluctuations in the DC Booster output. Since you have disconnected L2 it won’t be related to any of the other resistors or capacitors on the PCB.

The only three components that take power from this line will be the op amp, the 3.3V regulator on the sd card PCB, and the Arduino itself. Are any of these hot?

If the op amp and the 3.3V regulator look fine, there’s a chance that it’s an internal issue with the Arduino. Perhaps it’s able to produce 4.6V, however maybe not able to pump out the current needed. Let’s hope this isn’t the issue — replacing Arduinos aren’t worth the time. If you think the Arduino is the culprit, let me know and I’ll send you a new PCB with an Arduino in the mail.

Spencer

theostoelzl, 2019-01-15:
[image]
[image]

I haven't connected the SD PCB yet, so I don't know about the 3.3V regulator. And the op amp gets slightly warm at best, I don't think you could even remotely speak of hot there.

I highly appreciate the offer but in case the Arduino is the issue I have four spare PCBs from Elecrow, so that really won't be necessary!

On a side note, I just connected the display and noticed that it didn't work either. (The VCC pin is connected correctly.) Plus, sometimes the LED flashes randomly. Could those be related to this issue?

Thanks a lot,
Theo

spenceraxani, 2019-01-15:
Hi,

The detector will work fine without the SD PCB. We reference the Arduino  to 3.3V with the builtin regulator on the Arduino. R17 connects the reference pin to the 3.3V pin. 

The op amp should not get warm in the slightest. It’s a good bet that you have a problem in this area. I haven’t seen an op amp break, but I have seen many many times improper soldering joints on the op amp. Try retouching those with the soldering iron and making sure you have a good connection. You could alternatively remove the op amp, and see if the voltage on the booster goes to 29.5V.

The OLED should work independently of the rest of the circuit. The 4 pins on the OLED go straight to the Arduino. Have you uploaded the OLED.ino code to the Arduino?

The LED might be all over the place if the SiPM PCB isn’t plugged in. Don’t worry about this until the voltage issue is solved. 

Thanks,

Spencer

theostoelzl, 2019-01-19:
Hi,

I've touched up all the solder joints on the bottom side of the PCB, again, and put some new solder on the op amp legs, and yet again nothing changed. I'll try removing it, next.

The OLED.ino is uploaded; however, with the 328P old bootloader preset (if that changes anything, it wouldn't let me upload otherwise). The display remains black.

Thanks,
Theo

dnbaldwin, 2021-04-11:
Hello, Ive been following this with interest as I have built two boards now and both have exactly the same problem, with the U1 and L1 getting extremely hot immediately. I note that the board circuitry in the photos is a slightly  different version to the one for which pcb plates can be obtained. for example, mine has a U7, R18 and R 19 which are NS. These are not on the website version. Could I ask if anyone else has had this over heating problem, and if so how they resolved it? cheers, Dave