spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #35: Beautiful waveform through BNC, no joy through TP1 and 3

jimgallatin, 2019-01-06:
Spencer,

Thanks to you and others I now have a fantastic waveform through the BNC, looking just right.  A huge sense of accomplishment.  I also have a functioning OLED and LED.  However,

1.  I get nothing through TP1 (seems impossible since TP1 is a direct connection)

2.  I get a wave through TP2 which pulses in tandem with the BNC wave but which is much smaller than the BNC wave (seemingly the opposite of being amplified); this may be some glitch in how I have set up Channel 2 in the scope.

3.  I get nothing through TP3

4.  The OLED shows no counts and the LED doesn't flash

I had no glitches in loading the Arduino software and the splash screen works fine.  I have quadruple checked every solder joint.

If I understand the debugging notes in the manual this points to a problem with the Arduino software?  If so, I suppose I could cut and paste the source code from Github, compile and reload it?

Happy New Year down there,

Jim Gallatin

spenceraxani, 2019-01-06:
Hi Jim,

Glad to hear you are seeing the correct waveform on the BNC. You likely have a problem with the peak detector part of the circuit.

It seems very strange that the TP1 doesn’t show anything since it is almost the exact same connection as the BNC. One possibility is that your scope problem is 10X rather than 1X and it is scaling the size of the voltage by a factor of 10. 

TP2 should be approximately 20x larger than the BNC pulse. You might be seeing a small pulse again due to the scope probe. Or, it sounds like you might need to just scale channel 2 on the oscilloscope properly. Keep the time divisions the same as channel 1, and make the voltage divisions about 200mV. I would keep the trigger set to channel 1, to trigger on the BNC pulse.

If it turns out that you are getting the proper pulses on TP2, and nothing on TP3, then your next issue is with the peak detector circuit. The most common thing I’ve seen here, is either a bridged leg on the op amp, or a rotated diode — diodes have direction. I’ve also seen people put in 10uF caps into C6 rather than 10nF.

You don’t have to ever cut and past the GitHub code. Simply download the code, open it with the Arduino IDE, and hit upload. It sounds like the Arduino code is working fine though.

Thanks,

Spencer

> On Jan 5, 2019, at 8:35 PM, jimgallatin <notifications@github.com> wrote:
> 
> rough TP2 which pulses in tandem with the BNC wave but which is much smaller than the BNC wave (seemingly the opposite of being amplified); this

spenceraxani, 2019-01-06:
> Spencer,
>
> Your patience has paid off. I have a working detector. I am really pleased.
>
> I had the D1 diode reversed. Great call by you. The TP misreadings were due to my gross incompetence with adjusting an oscilloscope.
>
> The detector is counting away. Once school starts I will connect to my professor’s Python, download data, and see what I have. I have some air travel coming up and will bring the detector along and connect it to a USB battery and see how that goes.
>
> I have a ways to go before I can claim final success but it is a great feeling to have the device up and running.
>
> When I get some time I will post a list of my rookie mistakes so others might learn from them. I think I made every one in the book.
>
> Great way to start my New Year,
>
> Jim

Glad to hear the detector is working!

closed 2019-01-09 by jimgallatin