spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #36: Flying with a CosmicWatch detector

Pappy85, 2019-01-08:
The CosmicWatch paper has a really nice plot of count rates during a flight. I'm wondering how many people have tried this measurement and if they'e experienced any problems getting the detector through security. Any notes about your experiences would be greatly appreciated!

spenceraxani, 2019-01-08:
Hi,

I’m not sure how many people have taken the detectors through airport security, but I typically take a few detectors with me wherever I travel. Over the past few years, I’ve probably taken them on well over 50 flights and have never ran into issues. The TSA is more concerned with the cell phone power banks that I use to power the detector, since there are airline regulations limiting the size of Li ion batteries.  The detector is essentially nothing but a circuit board and a piece of plastic and I think the TSA is trained well enough to understand that it’s not something of interest. Sometimes in air, the flight attendants will ask about the detectors, however it’s mostly out of curiosity (I think many of them know that they are subjected to an increased amount of radiation). Or, sometimes they want to ensure that it doesn’t transmit any signals.

I would like to say that this was my personal experience and I can’t speak for the TSA on the subject. I wouldn’t want someone to find themselves in trouble because of this. If you are concerned about whether or not you can bring it on the plane, I believe you are able to contact the airport to speak with a TSA agent. 

Hope this helps! I’ve attached an image of my most recent flight to the South Pole below that’s pretty neat.

Thanks,

Spencer

Pappy85, 2019-01-09:
Thanks for the insight Spencer. 50 flights and no problems is very reassuring. I'm looking forward to trying it on my own. And thanks for sharing the detector design and supporting those of us building them. Its much appreciated!

closed 2019-01-09 by Pappy85

jimgallatin, 2019-02-14:
Hey all, I want to repay Spencer and everyone's help by advising that I took and used my detector on a flight from the east coast without issue.  TSA didn't ask about it, nor did the flight attendants.  I plugged it into my Mac for the first half of my flight, getting counts up 22 Hz around 30,000 feet.  I recorded the data using Python but haven't looked at it yet.

Bottom line, I flew using it without incident.

Jim