spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #37: Count

jimgallatin, 2019-01-11:
After a week of operating my Detector reads .2847 with +/- .001.  I didn’t see count guidelines in the repository materials (I easily could have missed it given my track record in building the Detector).  Are these readings about right?  Haven’t had the chance to download the raw data to my Mac.

spenceraxani, 2019-01-11:
Hi Jim,

If everything was built properly, then you should expect to see roughly 1Hz. This will depend on the environment. Areas with higher levels of radioactive backgrounds might have 1.3Hz, and areas with little background might be 0.7Hz. 

0.2847Hz sounds very low to me. Unless it is connected to another detector operating in coincidence mode. Then it sounds about right if the two detectors are sandwiched together. The low count rate could be due to an issue with the circuit, the software threshold in the Arduino code (if it was changed), or the construction of the detector. I doubt it’s due to the construction, unless you aren’t using any optical gel or reflective foil.

Thanks,

Spencer

> On Jan 11, 2019, at 8:41 AM, jimgallatin <notifications@github.com> wrote:
> 
> t? Ha

jimgallatin, 2019-01-11:
Hey Spencer,

Thanks.  

It is a single detector. 

I used optical gel and tinfoil.  (I could try wrapping another layer of tinfoil around the scintillator/SiPM combo.)  

I uploaded the code unmodified in any way.

The detector is currently in my basement which is under a concrete first floor.  I will move it to the first floor and see if the count rate improves.

Then I will go over the circuit again.

Jim

jimgallatin, 2019-01-13:
Hey Spencer,

I moved the detector to the first floor (above the concrete base of my house).

I checked the threshold in the Arduino code.  It was set at 50.  I changed it to 30 and uploaded the modified code.

I am now getting a count rate of 0.422, +/-.012, after about an hour.  It started out around 0.500 but has stabilized for now at 0.422.  I'll check it again after 24 hours.

Jim

spenceraxani, 2019-01-14:
Hi Jim,

It’s likely that your single floor house has ~6inches of concrete between levels, which will not affect the muon rate in any noticeable way. This will however change the comic electron/positron flux by roughly a factor of 3. When you decrease the threshold, you will be letting in more background (primarily gamma radiation). See, for example, the green histogram in Fig. 13 (https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/blob/master/ThePhysicsPaper.pdf). 

I still think there is an issue with your detector that is causing the count rate to be so low. An oscilloscope will help trouble shoot. Connect up to the BNC connection, and wait for a pulse. If your threshold on the oscilloscope is ~15mV, your detector should flash the LED on just about ever pulse that you see on the oscilloscope. 

It is my guess that you are close though. Perhaps it’s just a misplaced resistor causing a lower level of amplification. 

Thanks,

Spencer

> On Jan 12, 2019, at 6:00 PM, jimgallatin <notifications@github.com> wrote:
> 
> d

jimgallatin, 2019-01-14:
Thanks again.  I think I should make sure my incompetence with an oscilloscope isn’t getting in the way.

Trigger set at 15mv, edge, rising, auto mode
Horizontal set at 500ns, main window, trigger hold off 500ns
Vertical set at 10mv increments, probe 1x, coupling AC
Acquire is in sample mode
Measurements Pk-Pk 3.2 to 50.0+ mv, rise time 9-500 ms

Wave looks like specified

I am getting a flash for each trigger, but am also getting more flashes than triggers, which I assume is ok.

My triggers waves look just like your materials in shape.

Assuming my scope readings are ok, I will move on to re-examining each resistor in the amplifier circuit as you suggest.

I’m now running around 0.440 hz 

Jim

jimgallatin, 2019-01-14:
I forgot to mention I have a solar roof covering all of one half of my roof.  Would that impact the count?

jimgallatin, 2019-01-14:
Never mind.  No difference outdoors not under the roof.

jimgallatin, 2019-01-15:
Hey Spencer,

Given my ineptitude with a scope, and the fact it, getting counts, I think I will replace the resistors around the amplifier circuit (I installed the right ones but one may be bad or badly solder3d in a way 8 can’t see).  Then I will build my final detector using everything I have learned, see if it works well, and move to studying the circuit design and underlying particle analyses in your excellent paper (which I didn’t see before).

There is no end to the things I am and can keep learning from this project.

Jim

spenceraxani, 2019-01-15:
Hi Jim,

Glad to hear it. The second detector always goes much quicker and much smoother than the first. 

Thanks,

Spencer

jimgallatin, 2019-01-17:
Spencer,

I replaced the three resistors (R3, 5, and 20) in the Amplifier portion of the circuit.  I’m still hovering around .430 Hz.

Any final suggestions before I move on?  Replace the main op-amp component?  I’d really like to have two detectors in the end running together.

I am very confident I used the right tinfoil and tape, and I have triple-checked every solder joint, so I am assuming that I have a “weak” component.

Jim

spenceraxani, 2019-01-18:
> PS, I don’t know enough about electrical circuits to figure out on my own which would be the most likely to suppress the amplification. As I mentioned earlier, when I use the BNC the LED is flashing for every wave and even when there is no Wayne, so I am chalking that up to poor scope skills, and focusing on the low amplification.

Hi Jim,

On your scope, you should be able to set the trigger threshold to 15mV, and set the trigger from ‘Auto’ to ’Normal’ — this will cause the scope to update on every trigger.

Replacing the opamp won’t help either. The components that you purchased will come with an uncertainty in the values that they are suppose to be, and would all be negligible small compared to the change in rate that you are seeing.

I think starting your new detector is a good bet than come back to this later. It could simply be that the optical connection between the SiPM and the scintillator isn’t ideal and there are just more photons being lost. Or, potentially the light yield from your scintillator is just lower. Hard to say, but building your second detector might help out.

Thanks,

Spencer

jimgallatin, 2019-01-18:
Spencer,

Thank you.  Your comments are super helpful and give me more confidence in what I have done so far.  I will move forward with another detector.  

The project materials are absolutely right that this project exposes one to a world of related areas of knowledge, which is really terrific.  I find that I have to focus on successfully building the Detector or I will get sidetracked in exploring every area that is involved in the project, from how the circuit design works electrically to the processing of the initial photons.  Just reading the paper you linked me to forces me to make a list of each new concept that I don’t completely understand, and there are a lot of those for me to explore.

I am also starting to build something called the Radio Jove Project so the Cosmic Watch Detector experience will be invaluable there.

I will followup once I have the new detector up and running.

Jim

closed 2019-01-26 by jimgallatin