spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #40: SD Card Sockets

Johnkurveen, 2019-01-26:
Hi, I'm just getting into this project, and I have a quick question about the SD card sockets.

What are the requirements for the SD card sockets? You link to a $12 purchase on Amazon, but since we are only making one, it's more economical to purchase an individual model from Digikey. 

Here is a selection of SD card sockets from digikey. As you see, there's quite a few.
https://www.digikey.com/en/supplier-centers/m/molex/microsd-memory-card-connectors

Will any of these work, or is there a significant difference? In the video of how to make the main and SD PCB boards, you said you ordered the wrong ones, but didn't explain what was wrong with them.

spenceraxani, 2019-01-27:
Hi,

There’s a few different form factors for SD card sockets. I designed the boards based off the sockets that I found were most economical on Amazon and Ebay. It doesn’t quite look like the any of the ones on Digikey are the correct size though. It’s too bad there are so many different versions of this component.

Thanks,

Spencer

Johnkurveen, 2019-01-28:
Hi Spencer,

I think you've done a great job designing this project to be cheap and easily made, and I think one way to improve on these qualities is if we can find a way to use more SD card sockets. I know the shape and size of each socket is different, but the pinouts are in the same order at least. Are their size differences for the pins, or is it something else which makes different sockets incompatible?

 John