spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #42: Information on Arduino code

DeeTech, 2019-03-28:
Hello Sir,

I wanted to know about the following lines in code..

"const long double cal[] = {-9.085681659276021e-27, 4.6790804314609205e-23, -1.0317125207013292e-19,
  1.2741066484319192e-16, -9.684460759517656e-14, 4.6937937442284284e-11, -1.4553498837275352e-08,
2.8216624998078298e-06, -0.000323032620672037, 0.019538631135788468, -0.3774384056850066, 12.324891083404246};"

How you decided these values & how to calculate them..
I am confuse as to how to explain these values...

Thanks.

spenceraxani, 2019-03-28:
Hi Deepak,

These are the numbers used to convert a measured ADC value to a SiPM peak voltage. They correspond to the coefficients of a fitted 11th order polynomial. The fit is shown here:
https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/blob/master/Pictures/callibration.pdf <https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/blob/master/Pictures/callibration.pdf>

and described in Sec. 5.2 here:
https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/blob/master/Instructions.pdf <https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/blob/master/Instructions.pdf>

Essentially, we just inject a pulse that looks like a SiPM pulse (from an arbitrary waveform generator) of known amplitude and have the Arduino report a measured ADC voltage. 

Hope this helps,

Thanks,

Spencer

DeeTech, 2019-03-28:
Hello,
1st Thank you for reply.
I understand your point & I read above ref.

But wgat I dont understand is how this value come or calculated..

like  "-9.085681659276021e-27"  & others ..

did it display on arduino after you inject pulses & measured ADC?

You calculated it & then put all 12 values in array cal[];???

Thanks.

spenceraxani, 2019-03-28:
In the plot showing the measured ADC values versus SiPM Peak voltage, the SiPM peak voltage is controlled by setting the size of the input pulse on the waveform generator. The detector itself measures how large the pulse is in units of ADC counts and sends the information through the serial connector on the Arduino. We send ~500 pulses at a given size to quantify the uncertainty in the reported ADC value. The plot shows (in black) the mean ADC value as well as the spread in the reported values. Now that we have these datapoint, we fit an 11th order polynomial to the data — so, for example, one of the reported coefficients on the polynomial is -9.085681659276021e-27. After fitting to the data, we then use that fit in the Arduino code to convert measured ADC values from events, to original SiPM voltages.

Hope this helps.

Spencer

DeeTech, 2019-03-29:
Hello,
Thanks.
I understand theory point of view.
But????

As you said above "values in cal[] are coefficients of 11th order polynomial, that means
they are values of a0,a1....a11 in  this right? " "f(x) = 1/1-x = a0+a1x^1+a2x^2+....a11x^11  ?????

So I want to know these values like -9.085681659276021e-27 are displayed on arduino or arbitory generator to put in cal[ ] ??
Or did you use some software or something to get these values...

 because these values are +ve & -ve & increasing in exponential like e-27 then e-23 then so on & last value is ~12. which is also written as 12.324891083404246 . SO i want to know how these values come? Or how you calculated? or Where these values displayed? Or did you just do some maths to get these values & put them in cal [] array?

&
 inject ~200 pulses then calculate the standard deviation of the measured ADC
values.
What is this? & how to calculate it?
Is this related to the values getting in cal[ ]?

Thanks.

Deepak.

spenceraxani, 2019-03-30:
Hi Deepak,

> 
> Hello,
> Thanks.
> I understand theory point of view.
> But????
> 
> As you said above "values in cal[] are coefficients of 11th order polynomial, that means
> they are values of a0,a1....a11 in this right? " "f(x) = 1/1-x = a0+a1x^1+a2x^2+....a11x^11 ?????
This is correct.
> So I want to know these values like -9.085681659276021e-27 are displayed on arduino or arbitory generator to put in cal[ ] ??
> Or did you use some software or something to get these values...
The detector prints out the measured ADC value to the serial port when it triggers on an event. So, I send in hundreds of pulses of known amplitude and read the reported ADC values on my computer through the USB connection. In this case, I actually just read the values through the builtin serial monitor on the Arduino IDE. The arbitrary waveform generator just allows me to set the size of the pulse that I am injecting.  I then plot the injected SiPM pulse peak voltage versus the measured ADC value. 
> because these values are +ve & -ve & increasing in exponential like e-27 then e-23 then so on & last value is ~12. which is also written as 12.324891083404246 . SO i want to know how these values [...]
I fit the data gathered in the previous step using python. Is that what you are asking? Specifically, I use the Scipy package to fit the data accounting for the uncertainty in the measurements. Scipy reports the fit values (the coefficients of the 11th order polynomial), and I insert them by hand into the cal[] array. This is a calibration of the electronics. If the electronics change, then this fit has to be re-performed. 

If you plot the 11th order polynomial f(x) = a0+a1x^1+a2x^2+....a11x^11, with a0 - a11 from the cal[] array, you would get the blue line in the calibration plot that I sent in the previous email.
> &
> inject ~200 pulses then calculate the standard deviation of the measured ADC
> values.
The standard deviation describes the spread in the measured values.  
https://en.wikipedia.org/wiki/Standard_deviation
> What is this? & how to calculate it?
> Is this related to the values getting in cal[ ]?
Hope this helps,

Spencer

DeeTech, 2019-03-31:
Hello,
 
"If you plot the 11th order polynomial f(x) = a0+a1x^1+a2x^2+....a11x^11, with a0 - a11 from the cal[] array, you would get the blue line in the calibration plot that I sent in the previous email."

So this is how you get those values in cal []...
I understand now.

Thank you very much for your time.
I sincerely appreciate it @spenceraxani 

Regards.
Deepak.