spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #43: Slave has higher count than master

bojohnson02, 2019-04-10:
We have 2 detectors which seem to work by themselves. But when used in coincidence mode, the slave detects even more than the master. We're not sure what would cause this issue, but were wondering if it's been seen before.

closed 2019-04-10 by bojohnson02

spenceraxani, 2019-04-10:
Hi,

There’s likely a problem. By definition, this should never happen. I haven’t seen this before — what sort of count rates are you seeing?

If the D6 pin on the master detector is HIGH, then the coincidence detector will report every trigger it sees as a coincidence. It could be a short in your 3.5mm connector, can you check your connections? If the power connection on the 3.5mm connector is bridged with the coincidence connection, it could cause your problem.

Thanks,

Spencer

bojohnson02, 2019-04-10:
It looks like we just had the 3.5mm connectors not in all the way because the face plates in the back. We were getting rate counts for the master at about 1.5 with the slave at about 1.7