spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #44: Multiple Issues

rptamin, 2019-04-19:
Hello Spencer,

We are attempting to put this detector on a rocket at Spaceport America Cup, and see the change based on altitude as our scientific payload. However, we are having quite a few issues with the detector.

1. There are time stamps for only some measurements. We are not sure why this occurs. What are ways we can ensure there is a timestamp for all measurements
[image]

2. The measurements are not consistent and do not seem to make sense. The only way we get measurements is if we press down on the detector. Even then, it is very inconsistent. I replaced the 6 pin header, however, the problem persisted. The ADC values also seem to be off, they are only ranging from 0~10.

Thanks,
PSP Solids Team

spenceraxani, 2019-04-19:
Hi PSP Solids Team,

The data has me confused. How did you record his data (computer, SDCard, serial monitor)? Are you using the executable for windows or the python code? I would strongly suggest using the python code.

Can you link me the GitHub you are working out of?

When plugged into a computer and recording data using the import_data.py script, you should get a time and date stamp by the computer. It looks like you do see a time stamp (23:xx:xx.xxx), however I do not see the date.

I also see arrows, which is an indication that the script is not parsing things correctly. If this is the case, it’s my guess that you have a different library version of one of the programs.

Can you use an SDCard to see if the data is looking alright? When you just power the detector, does the OLED screen report reasonable values (count rates)?

I also see that on the events that do not have a time stamp, that the ADC value is 0 or 1023. These two values are the two extrema values it could have. Zero should never occur, since we set a threshold at 30 ADC counts. Did you modify the Arduino code at all?

Thanks,

Spencer

rptamin, 2019-04-19:
Hi Spencer,

So some good news! We were able to fix the problem of problem of it having to be pressed down upon. This was due to the SIPM not having a good enough connection. We desoldered the sensor, and re soldered using solder paste and hot air gun.  So we think we solved our hardware problem, however, our software problem still exists.
[image]
This is what we get now. The threshold is 50 and it appeared at approximately .3 hz. We will try to use the python script and make sure we get time stamps.

Thanks,
PSP Solids Team