spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #45: No Detection Of Muons

Aquastar1017, 2019-05-05:
Hey Spencer,

I'll try and upload pictures in two days if I can as the project is currently in the lab, but I will explain the problem as best as I can.

Currently, the muon detector does not detect any muons. Well, technically, it usually detects exactly one when it is reset or receives voltage. When we try and give the main PCB an artificial wave pulse, we do "detect" muons, but when the SiPM is put in place, it results in only the one detection.  

When we used a multimeter on the VH going to the SiPM, it results in 29.66 V, which is in line with what it should. and so we thought it was a problem with the SiPM PCB.  We redid the SiPM to make sure it wasn't a problem with any of the components but it still results in the SiPM component heating up, which is probably not a good thing, and no detentions. 

There was a point at which C6 was not populated correctly, but we didn't think that that was a major problem.

Another note is that we are currently lacking the Temp sensor as we were trying to keep the budget small but will order one if absolutely needed.

spenceraxani, 2019-05-06:
Hi,

Seeing only a single detection is an indication that there is a problem.

It sounds like you tested the Main board using a waveform generator and are seeing roughly the correct behavior. This should normally be checked by injecting a pulse that looks roughly like a SiPM pulse, and then checking test point 2 and 3 (TP2 and TP3). If these look good, then your problem is definitely with the SiPM PCB.

The SiPM should never be heating up. Could it be that the orientation of the SiPM is backwards (this is the biggest issue I tend to see)? The cathode of the SiPM (pin 1) should be connected to pad 1 on the SiPM PCB. A short on the resistor R12 could also cause issues. The simplest first step is to add new solder to each of the components on the SiPM PCB. The new flux in the solder will make sure you get a good connection.

Hopefully whatever caused the SiPM to heat up did not damage it. 

C6 is an important component for the peak detector circuit (and thus the actually detection by the Arduino), but would not affect the SiPM performance or the Amplification circuit.

Thanks,

Spencer

Aquastar1017, 2019-05-06:
Hey again, 

Just wanted to give you an update before this can be closed. We populated a new SiPM board with a new SiPM (wow they are expensive) and did an ohm check for the SiPM. We encased it to make it light resistant and were not actually expecting it to work and mostly as a last ditch effort. 

It worked. We don't know how it is now working as we triple checked all of the components on the board especially R12. We can only guess that it was shorting somewhere and when we made a new board it fixed the issue.

closed 2019-05-06 by Aquastar1017

spenceraxani, 2019-05-06:
Hi,

That’s interesting. I’m glad it worked out though. 

Spencer