spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #46: Connection to Computer -- Connect and Disconnect

melloyellowblue, 2019-07-10:
Ok weird problem here:

My arduino keeps connecting and disconnecting from my computer as I try to do anything with it, especially if the SiPM is connected via the 6-pin connector.

From what I can gather from arduino webpages, this means that the current draw may be too high, but they were powering a servo, and this circuit shouldn't have too much draw.

[https://forum.arduino.cc/index.php?topic=496581.0](https://forum.arduino.cc/index.php?topic=496581.0)

I'm assuming from a lack of GitHub issues on the topic that people generally have not been having problems with this. I am using a genuine arduino nano from arduino.cc because I am basically 0-7 with the cheaper arduinos (terrible luck but I guess that's just how $5 arduinos work).

Any suggestions on how to fix this issue or what may be causing it? I am running Windows 10 if that helps.

spenceraxani, 2019-07-10:
Hi,

It’s strange that you had issues with the cheaper Arduinos. I’ve purchased ~300 of them, and ran into problems with roughly 1-3 of them. Could be an issue with the manufacturer, or potentially with your OS (I’m not very familiar with Windows). Are you able to load the Arduino sketch to it? You could start with the example sketch in the Arduino IDE called blink to make sure it’s working. Does the Arduino connection issue only occur on your computer? Can you plug it into a USB power bank, or different computer?

Like you suggested, my guess is that there might be a short somewhere. A USB power meter can be useful in determining if this is the case:
https://www.amazon.com/Soondar-Charging-Concurrent-Real-time-Smartphone/dp/B00ORNOWZK/ref=sr_1_12?crid=3S6UC3GECJPZ&keywords=usb+current+meter&qid=1562774990&s=gateway&sprefix=usb+current%2Caps%2C142&sr=8-12 <https://www.amazon.com/Soondar-Charging-Concurrent-Real-time-Smartphone/dp/B00ORNOWZK/ref=sr_1_12?crid=3S6UC3GECJPZ&keywords=usb+current+meter&qid=1562774990&s=gateway&sprefix=usb+current,aps,142&sr=8-12>

Could you also send me a photo of you PCB? I might be able to identify the problem. Sadly, if you are seeing issues with plugging in the SiPM PCB, the issue might be on that board.

Thanks,

Spencer

melloyellowblue, 2019-07-11:
Hello Spencer!
Here are the pictures of the PCB:

[image]
[image]

I don't think it's an issue with the SiPM PCB since this problem happens regardless of whether or not the PCB is plugged in. I am also having problems with the plugged in Arduino not being recognized by the computer (only when connected to the rest of the circuit), but that's a separate issue.

I am able to upload code to the arduino, but only when it's not connected to the actual board. This same issue happens on all computers I have tried, both windows and linux.

spenceraxani, 2019-07-11:
Hi,

One issues I can spot is that your temperature sensor is mounted backwards. I’m not sure what effect this has on the Arduino though.

The other potential issue (likely not) is that I cannot see the solder joints on the bottom of the board connecting the Arduino to the PCB. Your soldering looks particularly good on the other connections though — did you use reflow? While you are rotating the temperature sensor, it’s probably a good idea to check all the other connections by simply adding some extra solder to some of the suspect connections.

Thanks,

Spencer

> On Jul 11, 2019, at 10:09 AM, melloyellowblue <notifications@github.com> wrote:
> 
> o uploa

melloyellowblue, 2019-07-11:
Hello!

I wanted to make sure I could still, in some capacity, connect to the arduino, so I intentionally have not soldered it to the PCB. I have found through making another PCB with a genuine arduino that it is impossible to connect and recover data directly to a computer, so I wanted to figure out the problem before I permanently attached the arduino to the board.

I will try switching the temperature sensor, and see if it has any effect.

Thanks for noticing the other joints! I did not use reflow soldering, just a combination of thinner solder (0.020") and getting too close to the PCB to be safe. Haven't gotten too many fumes in my lungs so far so ¯\_(ツ)_/¯. I found that it was pretty great with the smaller chip components.

melloyellowblue, 2019-07-11:
Update: Fixing the temperature sensor made everything work! I guess it might have been sending a negative voltage or something to V_out and causing the arduino to freak out. Thank you so much!

closed 2019-07-11 by melloyellowblue

spenceraxani, 2019-07-11:
Glad to hear that you got it working!

Spencer