spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #48: overheating board

AbbeySiegel, 2019-09-05:
Hello, I'm having an issue where part U2 has been overheating. I have been able to create a full circuit that gets the correct voltage but after a couple minutes of running this part will overheat to the point where it burnt to the touch. Have you had this issue come up before? If so is there a way to fix it? I replaced the part itself and the issue continued to happen. 

Thank you,
Abbey Siegel

spenceraxani, 2019-09-05:
Hi Abbey,

My best guess is that the op amp is mounted facing the wrong direction. Can you double check that pin1 is connected to the pad with the dot beside it? You can identify the side of the op amp with pin 1 by looking at the edge that has a bevel on it.

It definitely should not be generating any noticeable heat. You can also send me a photo of the board and I might be able to identify the issue.

Thanks,

Spencer

AbbeySiegel, 2019-09-05:
Here is a picture of the board. I’m going to try to flip around the op amp and see if that helps. Thank you! 

Abbey Siegel

spenceraxani, 2019-09-05:
I’m not sure if the attached file worked.

AbbeySiegel, 2019-09-05:
I’ll try again. I don’t know why it wasn’t working the first time.