spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #49: Event rates reported when connected to MIT server

pedelen, 2019-09-05:
Hi Spencer,

I have built several devices and they all report slightly different rates which isn't surprising. I have noticed that when running in coincidence mode, the rate reported when connected to the MIT server is slightly lower than that displayed on the OLED or saved to the MicroSD card. This occurs when running the systems with a windows PC or when running a Linux machine. Any thoughts?
Thank you, Paul Edelen

spenceraxani, 2019-09-05:
Hi Paul,

Sorry about that. It's actually because the website doesn’t account for the dead time of the detector correctly. It is on our list of things to fix (hopefully soon). 

Thanks,

Spencer

pedelen, 2019-09-05:
Hi Spencer,

 

Thank you, I appreciate the feedback.

 

Paul

pedelen, 2019-12-22:
Hi Spencer,
Has anyone used several detectors in coincidence mode with a liquid scintillator sandwiched in the middle to attempt a rough measurement of muon mass?

closed 2019-12-23 by pedelen