spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #51: Regarding calibration process

Rithishbarath, 2019-09-24:
Dear Axani
1. How to do the calibration part?

To determine the correlation between a measured ADC value from the peak detector circuit and the actual SiPM pulse amplitude, we begin with taking a screenshot of the SiPM output waveform, digitize it on a computer, and then program it into an arbitrary waveform generator. The waveform generator allows us to scale the voltage and inject it into the circuit through the BNC connection (without the SiPM PCB connected). The plot below shows the relationship between the amplitude of the injected pulse and the measured ADC value. In this image, at each point, we inject 200 pulses then calculate the standard deviation of the measured ADC values. We then fit the distribution according to a high-order polynomial. The result of the fit and the data is shown in Fig. 10. We use these results in the function “Calibrationfit” in the Arduino code.

Actually, my doubt is after going through this passage I find difficulty in digitizing it. How to get the screenshot of the SIPM output? Is there any software to digitize the output waveform? if so what is the name of the software? Is there a software which will generate the arbitrary waveform. Or do you have any standard waveform in the repository which we can directly use it? Does this calibration fit function has all these things worked out? or should we do this process for the detector we built? Is this process entirely different for each individual detector? Kindly give me a way out

2. At what temperature did you heat polish the scintillator?

3. Is there any Arduino script which can actually get both the sd card and OLED screen to function simultaneously? 

Address these queries as soon as possible so that we can continue from this point

spenceraxani, 2019-09-24:
Hi,

(1) Just to clarify: you do not need to recalibrate once you build a detector. The calibration should be very close to the one provided if you use the same electronics described on the GitHub. If the electronics change (say, changing the gain on the detector), then recalibration is needed (if you want an accurate description of the SiPM pulse in mV).

If you still feel you need to recalibrate, there exists free software online for extracting (x,y) coordinates from a graph. For example, plotdigitizer allows you to input an image, define the scale on the axis, then point click the (x,y) coordinates you want. 
http://plotdigitizer.sourceforge.net/ <http://plotdigitizer.sourceforge.net/>

The arbitrary waveform generate is a piece of hardware that lets you generate a specific voltage. These can be rather expensive though.

(2) We typically use the hottest temperature, with the lowest flow rate, on a soldering station. This tends to be somewhere around 450C.

(3) There exists Arduino code out there that has both the sd card and OLED scripts included. For example:

This code was made by someone else working on the project, however, I have not been able to test it yet. 

Thanks,

Spencer

Rithishbarath, 2019-09-24:
Thank you for the clarification. Which file has this calibration or
so-called Arduino code in the repository?

Adding to that we are actually finding a way to measure the lifetime of the
muon from this experimental setup. My query is simple, is there a way to
measure the lifetime from the data provided by this detector?

Rithishbarath, 2019-09-24:
> Hi,
>
> The OLED and SDCard Arduino code is located in the Arduino/ folder in the
> repository:
> https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 <
> https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2>
>
> It’s in the cal[] array:
> const long double cal[] = {-9.085681659276021e-27, 4.6790804314609205e-23,
> -1.0317125207013292e-19,
> 1.2741066484319192e-16, -9.684460759517656e-14, 4.6937937442284284e-11,
> -1.4553498837275352e-08,
> 2.8216624998078298e-06, -0.000323032620672037, 0.019538631135788468,
> -0.3774384056850066, 12.324891083404246};
>
> This set of numbers describes the coefficients of an 11th order
> polynomial. If you were to plot it, you would fin the blue curve below
> (calibration.pdf plot in the Pictures folder of the repo).
>
> You won’t be able to measure the muon lifetime with this detector since it
> is just too slow. The maximum sampling speed of the Arduino is roughly 8us,
> and you are looking for signals that are ~2us apart. Beyond this, the
> scintillator is rather small to make such a measurement.
>
> Thanks,
>
> Spencer

Thank you soo much axani.This project seems to be an exciting one and its
really helping me to apply for graduate programmes at USA.The experience i
gained while trying to build this detector is a bliss.Thanking you.

Rithishbarath, 2019-09-25:
Dear spencer,
                    Can you kindly let me know the drivers and Softwares
used to load this Arduino code into detector by windows 7?