spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #54: Regarding arduino soldering and screen

Rithishbarath, 2019-11-22:
[1. Kindly tell me whether the screen is faulty by looking at the attached image because I cannot see all the details or problem with the Arduino board

2. You have left two pins unsoldered in the main PCB (ARDUINO NANO). Should we do that too or should I use those pins?

Thank you](url
[image]
[image]

)

metapulp, 2020-01-03:
Hello, I have made several of these boards. I actually do not even solder the Nano onto the main Cosmic Watch board anymore, because the pins are snug enough. I solder the full length of header pins to the Nanos though then just push the Nano into place. I had also asked Axani for what solder he used, which he posted elsewhere on Github. This was some time ago, but it solved my soldering problems.

spenceraxani, 2020-01-03:
Hi,

I would strongly recommend soldering the Nano to the board. Most of the pins are used, and if any of them are loose, or become loose, then the detector will not function properly. 

I use this solder:
https://www.amazon.com/Kester-24-6337-0027-Solder-Alloy-Diameter/dp/B0149K4JTY/ref=sr_1_4?ie=UTF8&qid=1531342218&sr=8-4&keywords=solder+kester <https://www.amazon.com/Kester-24-6337-0027-Solder-Alloy-Diameter/dp/B0149K4JTY/ref=sr_1_4?ie=UTF8&qid=1531342218&sr=8-4&keywords=solder+kester>

Thanks,

Spencer

spenceraxani, 2020-01-03:
Hi,

There definitely appears to be an issue with the screen. The two pins left unsoldered on my boards are simply to reuse those header pins for the SDCard PCB, but it is definitely not required to remove them. 

My bet is that your screen is malfunctioning rather than the detector. I’ve gone through a few hundred screens and probably 10% of them have had some sort of problem (not this one though). The failure rate of the screen is definitely correlated with the cost of the screen; cheap ones from abroad have a much higher failure rate. I have been able to repair some of them though. If you look at the backside of the screen, there is a thick ribbon connection. I’ve found that sometimes the ribbon connection is not mounted properly on the screen’s PCB. This can be fixed using a hot air gun and repositioning the ribbon cable.

Hope this helps. 

Thanks,

Spencer

vcbomba, 2020-01-04:
Good morning Spencer, Do you have a source for the CW Muon detector PC board? The GetHub site source listing didn’t respond. Looking to build ten devices. Wanting to share the build with several of my grandchildren.

Thank you,

Vince

spenceraxani, 2020-01-04:
Hi Vince,

I typically use Elecrow.com for manufacturing the PCBs. I ordered some things from them the other day, so they should respond. You simply have to upload the PCB_Gerber_files.zip file from the Github repository:
https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/tree/master/PCB_Files <https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/tree/master/PCB_Files>

to this website:
https://www.elecrow.com/pcb-manufacturing.html <https://www.elecrow.com/pcb-manufacturing.html>

and select:
Different PCB Design = 10

Hope this helps.,

Spencer

vcbomba, 2020-01-05:
Spencer, I will do as you suggested. I appreciate the quick response. 

Thank you, 

Vincent C. Bombardier 
Senior Construction Manager
Walmart Construction/Store Planning
Region 40 & 43 
Mobile: 612-412-8400
Walmart Inc.

vcbomba, 2020-01-05:
Spencer, the Elecrow PCB site referencing the different PCB design goes from 1-9. 10 isn’t available. Please advise. 
- board dimension is set to 100mmX100mm w/ 1.6 thickness 2 layer, HASL finish, no stencil or castellated holes.  Would this be correct? 

Thank you, 

Vincent C. Bombardier 
Senior Construction Manager
Walmart Construction/Store Planning
Region 40 & 43 
Mobile: 612-412-8400
Walmart Inc.

spenceraxani, 2020-01-05:
Hi,

The 1-9 refers to the number of different designs. We have a single 100mmx100mm design, so you’ll want to keep this at “1”. The number above that box, “5” in the attached image, is where you select how many PCBs you want manufactured. You were able to click on the “5” and select any increment you would like, however it looks like it is not working at the moment. It is likely a problem with their website.

I’ve email Elecrow to notify them of the problem. You could either order two sets of 5 now, or wait for them to fix the issue and select 10 (10 PCBs is actually normally the same cost as 5).

Thanks,

Spencer

Rithishbarath, 2020-01-10:
Which line of the code  should be changed so as to reduce the time difference in coincidence mode between master and slave counter? by default it is 30 microseconds.Which line of the code should be modified?

spenceraxani, 2020-01-10:
That’s a tricky one to see. Every command in the Arduino takes a certain number of clock cycles to complete. I read the analog pin3 as a timer for the coincidence:

      // Wait for ~8us
      analogRead(A3);

So if you wanted to add another 8us to the time window, I would have this line run again:

      // Wait for ~8us
      analogRead(A3);
	analogRead(A3);

Thanks,

Spencer

Rithishbarath, 2020-01-10:
Thanks spencer.Im actually bit confused with the kind of coupling compound
which i should use.Ive a picture of the compound.Kindly tell me wheather i
can use that as a coupling compound.

spenceraxani, 2020-01-10:
Do you mean the that you are confused about the optical coupling? Something like this would be used for lab coupling:
https://www.ebay.com/itm/Silicone-Optical-Coupling-Compound-for-PMT-Photomultiplier-Scintillator-Detector-/261918290249 <https://www.ebay.com/itm/Silicone-Optical-Coupling-Compound-for-PMT-Photomultiplier-Scintillator-Detector-/261918290249>

I’ve also heard about using clear vaseline, however I haven’t tested it. I also do not see a picture attached to your previous message.

Spencer

Rithishbarath, 2020-01-10:
Dear spencer

Follow this link.Can i use this as a coupling  compound? Need your advice

https://www.amazon.com/Loctite-29329-30Cl-Glass-Bonding/dp/B00BWEOW84

spenceraxani, 2020-01-10:
I’m not sure whether or not it would work. If you do use adhesive though, you won’t be able to remove the SiPM, which I’m sure you are aware of. I would test the adhesive on the scintillator and a piece of acrylic to see what the coupling looks like. Then decide if you want to use it. It also has to be transparent to ~400nm light.