spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #56: led blinks but Not getting any signal in all three dedicated pins.

AbidHussain70, 2020-03-09:
Hi, first wanna say thanks for such a beautiful and well documented project. 
I follow all of your steps and exactly copy your work. I did all the troubleshooting, DC-DC booster working fine, OLED working, SD card, and led also works. The problem is, that when I attached the SiPM PCB + plastic scintillator on main board the led does start blinking, but I am not getting any pulse on BNC or any of the three pins dedicated for checking pulses. I am confused, on one side no pulse while on the other side the led blinks. 
Can you help please.

Another thing I just missed one test, I did not check the plastic scintillator with torch. Which torch should we use for this purpose.
Looking for your answer thinks in advance.

spenceraxani, 2020-03-09:
Hello,

If you have measured the DC-DC booster output and are seeing 29.6V, but are not seeing any pulses on through the BNC connector, then it is an issue with the SiPM PCB. The most likely two candidates are that either the SiPM was mounted backwards, or perhaps the connection on one of the SiPM pins came loose while securing it to the scintillator. 

Seeing blinking seems odd though. If it is rhythmic, then perhaps you have the wrong code uploaded? Although that wouldn’t explain why you don’t see pulses from the BNC connection. It could be triggering on noise or spurious signals from the SiPM PCB, which could look random and not like SiPM pulses, this could be explained by a broken connection on the SIPM pin. Sadly, I think you’ll have to unwrap your scintillator to check it.

Thanks,

Spencer

AbidHussain70, 2020-03-10:
Thank you for quick response.
 """"........ If you have measured the DC-DC booster output and are seeing 29.6V, but are not seeing any pulses on through the BNC connector, then it is an issue with the SiPM PCB. The most likely two candidates are that either the SiPM was mounted backwards, or perhaps the connection on one of the SiPM pins came loose while securing it to the scintillator......."""

Yes DC-DC booster output is giving 29.65V.
[image]
For the second part of your comment, I checked it several time. I solder 2 SiPM PCBs, but same problem. 
Check the picture pin 1 of the SiPM has a small extra pin as you said in video 3 (https://youtu.be/yFgin5wlw4I).
[image]

And in the SiPM PCB pin one is labeled with small circle.
[image]

I solder the SiPM exactly in the same direction. And I also checked the resistance, in one direction it is giving some finite value (in Mega ohms) in other direction no value. Please check the pictures.

[image]

[image]

 
And from datasheet of the SiPM pin 1 is anode and pin 3 is cathode. 

[image]

I check the voltage drop on anode and cathode pin of the SiPM. Which was working.

[image]

 
 """"........ Seeing blinking seems odd though. If it is rhythmic, then perhaps you have the wrong code uploaded? Although that wouldn’t explain why you don’t see pulses from the BNC connection. It could be triggering on noise or spurious signals from the SiPM PCB, which could look random and not like SiPM pulses, this could be explained by a broken connection on the SIPM pin. Sadly, I think you’ll have to unwrap your scintillator to check it..........."""""

 Blinking in rhythm, well I can not say anything about this, but it does blink, sometimes first and sometimes slow.  I upload OLED code from v2. I unwrap it, in fact I did unwrap it several time.

spenceraxani, 2020-03-10:
Hi,

It actually sounds like your detector is working. Could it be that the issue is with the oscilloscope? Perhaps the horizontal center if off-scope, or something like this? Just to ensure you are looking for the correct signal, it should be a positive pulse, 10s of mV in voltage, and order 500ns in width. When you use the OLED screen, what is the approximate count rate that you see?

I forgot to reply to your question about the scintillator. I use a hot air gun from a soldering station to melt the scintillator. This isn’t really needed. If you have a good optical surface between the SiPM and scintillator, improving the reflection on the sides won’t dramatically help collect photons.

Thanks,

Spencer

AbidHussain70, 2020-04-14:
Sorry I forgot to close the issue. The problem was with SiPM soldering. Though I did troubleshooting as you mentioned in your video and in the document, the SiPM circuit was showing same results. But due to the SiPM tiny pins, it became disconnected somehow during installation of SiPM circuit on main board.  
I used Rosin Flux to fix the problem. Now the SiPM is solder properly and circuit is working fine.
Thank you.
[image]

[image]

.

closed 2020-04-14 by AbidHussain70