spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #58: SIMP or scintillator  problem

mriosmar, 2020-06-24:
Hi.
I have a problem with the SIPM module of the detector. I have assembled the SIPM module and apparently everything is fine:
1) The voltage between the HV and GND pins is 29.3 V.
2) The chip is installed correctly (pin 1 at the circular mark)
3) The vontage between PIN 3 and PIN 1, when the circuit is on is 25.7V
4) I have reviewed connections and solder points and everything is OK.
5) The connection between BNC and SIPM module is ok too.
6) I have wrapped the scintillator as indicated in the instructions.
But only one event appears on the screen.

If I connect the oscilloscope to the BNC port I don't see any pulse and neither on TP1. This shows that the problem is in the SIPM module.

I have tried a second SIPM module that I have mounted and with the same scintillator (I only have one) and the same thing happens.

Could the scintillator not be working properly? Is there a way to verify the scintillator?

This is a bit desperate because I have dedicated tens of hours to the project and the detector does not work.

Thank you very much

spenceraxani, 2020-06-24:
Hi,

It sounds like you are correct in interpreting the problem as an issue with the SiPM board. If you see the correct voltage, but do not see anything coming out of the BNC, then the issue is somewhere on the SiPM PCB. Can you send me an image of the front and back? I might be able to tell you where the issue is. 

Normally, I would have assumed that you have an issue with one of the SMT components on the SiPM PCB. But since you were able to get the DC-DC booster working (the hardest part of the soldering), and have completed two boards, I would say that it is unlikely the issue. Nevertheless, the most common issue is a non-properly soldered pad on one of the SMT components. 

It’s unlikely to be an issue with the scintillator — unless it’s not actually scintillator. Sadly, the best way to see if the scintillator is working is to use a SiPM PCB that is known to work.

Could it be an issue with the oscilloscope? The pulses should be upwards going, maybe 10s of mV in height, and 100s of ns in width. These should come in at ~Hz frequencies.

Thanks,

Spencer

mriosmar, 2020-06-24:
Hi.

I have measured the resistance (at the solder points) between pins 1 and 3 of the SIPM and the values ​​are 13.6Mohms and in the opposite direction 3.4 Mohms. This shows that the solder points appear ok. I have added the photos.

On the oscilloscope I have used the 20 to 100 mV scale to detect the pulses and on the 0.5 microsecond scale and I see a noise of about 10mV in amplitude but no major pulse.

Also if I modify the variables SIGNAL_THRESHOLD = 1 and RESET_THRESHOLD = 1 in the arduino program I can see in the console the measurements it makes of pin A0 of the Arduino and this is what I get:

23:39:17.239 -> ### Comp_date Comp_time Event Ardn_time[ms] ADC[0-1023] SiPM[mV] Deadtime[ms] Temp[C] Name
23:39:17.339 -> ##########################################################################################
23:39:17.393 -> Rios
23:39:18.342 -> 1 37 352 66.89 73 24.01
23:39:20.397 -> 2 2059 3 11.36 2166 24.44
23:39:21.345 -> 3 3034 3 11.36 3173 24.01
23:39:22.400 -> 4 4078 1 11.97 4252 24.34
23:39:23.401 -> 5 5046 3 11.36 5253 24.34
23:39:24.350 -> 6 6047 3 11.36 6286 24.01
23:39:25.406 -> 7 7061 6 10.70 7334 24.01
23:39:26.409 -> 8 8091 3 11.36 8398 24.44
23:39:27.358 -> 9 9048 6 10.70 9388 24.34
23:39:28.414 -> 10 10068 6 10.70 10441 24.44

This shows that no pulse greater than 11mV appears in the SIPM which is consistent with what I receive by BNC on the oscilloscope.
[image]
[image]
[image]

I bought the scintillator at the recommended place in the shopping list:
https://www.ebay.es/itm/263355405917?ul_noapp=true

I think something very strange is happening and I am unable to see it so far.

Thank you very much

spenceraxani, 2020-06-25:
It’s a bit tough to say for sure, but the SiPM does appear to be offset compared to the outline on the PCB. If it is too far off, it will short the connection with the large middle pin on the SiPM. I do not know what effect this would have, the datasheet simply says that this pin should be left floating.

Your oscilloscope settings are good. 

The first thing to try would be to disconnect the resistor R4 and look for pulses through the BNC out. This will ensure that the opamp downstream isn’t shorting the connection. 

Since you do not see any pulses out of the BNC connector, the Arduino isn’t doing anything but providing power to the SiPM. There is a DC offset on the peak detector circuit, and setting the signal threshold to 1 puts the trigger below the waveform. You can see that the triggers occur every second, this is because the OLED screen is being updated. 

Thanks,

Spencer

mriosmar, 2020-06-26:
Hi again.

I have fixed the position of the SIPM and I have ensured that it was well connected but I have not solved the problem.

Then I removed the R4 resistor to isolate the SIPM from the PCB but unfortunately I still don't get any pulse from the BNC port on the oscilloscope (with neither of the two SIPMs I have). I have checked the connectivity of the BNC port, the resistors and the SIPM. I wonder if any of the capacitors could fail. I have no way to verify it.

What role do capacitors C8, C9, C10, and C11 play in that circuit? Could this be the cause of the problem or does my question make no sense?

The other possible cause of the problem is the scintillator but I find it hard to believe that it could be the problem. It is the Bicron-BC408-Plastic-Scintillator bought on ebay.

My intention is to buy another scintillator to verify if it is the source of the problem but at the moment the seller does not send to Spain for the covid-19.

Do you have any idea to detect the source of the problem?

Thank you very much.

spenceraxani, 2020-06-26:
Hi,

The capacitors on the SiPM PCB are simply for filtering the power going to the SiPM (removing high-frequency noise on the 29.7V line). If any of them aren't connected, it shouldn't have a noticeable affect. However, if one of the is shorted (perhaps some solder is shorting the capacitor below the component)
, then you should expect problems. 

You can check this by measuring the resistance on the header pins, see attached image. Between the outer and middle sets, you should 100Ohms and 49.9Ohms.

Can you attach images of the Main PCB? It would also be good to double check that you get 29.7V+/-0.4V powering the board. An image of the wrapped up scintillator would also be a good idea.

Thanks,

Spencer

mriosmar, 2020-06-27:
Hi.

I have checked the intermediate points of the SIPM PCB and they are ok: 49.9 Ohms and 100 Ohms.

I have soldered R4 again and verified the voltage supplying the SIPM PCB. I get 29.3V (could it be a little low and be the source problem?).

I've gone back to review connections and solder points with a magnifying glass and apparently everything is fine.

I have components to mount a second Main PCB. I am going to install it in the next few days to rule out the Main PCB as the source of the problem.

Thanks again

[image]
[image]
[image]
[image]

spenceraxani, 2020-06-27:
Actually, my point was that removing R4 disconnects the input signal from the Main PCB. This would eliminate any interference from downstream components, like the op amp.

I think the issue might be with the solder on the BNC connection. The pictures look like it’s not properly soldered to the board. The two smaller connection are used for the signal, and should be checked.

mriosmar, 2020-06-28:
Hi.

Eureka!!!

My detector works! This is what I see on the port monitor:
13:06:34.988 -> ### CosmicWatch: The Desktop Muon Detector
13:06:35.042 -> ### Questions? saxani@mit.edu
13:06:35.088 -> ### Comp_date Comp_time Event Ardn_time[ms] ADC[0-1023] SiPM[mV] Deadtime[ms] Temp[C] Name
13:06:35.189 -> ##########################################################################################
13:06:35.242 -> Rios
13:06:37.873 -> 1 1678 473 126.29 109 27.34
13:06:38.613 -> 2 2439 167 28.76 150 27.77
13:06:40.599 -> 3 4426 65 18.22 224 28.10
13:06:44.375 -> 4 8179 113 23.02 365 27.34
13:06:44.460 -> 5 8278 177 30.25 369 27.34
13:06:47.568 -> 6 11367 103 22.23 478 27.34
13:06:48.088 -> 7 11897 48 15.29 482 28.10
13:06:48.317 -> 8 12116 96 21.67 520 28.31
13:06:50.106 -> 9 13888 207 35.26 558 27.56
13:06:50.676 -> 10 14462 124 23.92 597 28.10

The problem was that the scintillator was wrapped with a single layer of black tape. I have wrapped it with three layers and they have started working.

Spencer, thank you very much for your responses.

spenceraxani, 2020-06-28:
> 13:06:35.042 -> ### Questions? saxani@mit.edu <mailto:saxani@mit.edu>

That’s great! Glad to hear it.

closed 2020-06-28 by mriosmar