spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #61: Electronics Calibration

claudepi, 2020-11-24:
Hi Spencer, 

First off, let me say thank you so much for putting together this wonderful project. It is very impressive and I have already learned so much (though I am still very much an electronics novice). 

I am interested in doing an electronics calibration as you describe in Section 5.2 in Instructions.pdf. I do understand that this is not necessary - I simply want to do it as an exercise. After lots of trial and error and debugging, I have finally arrived at the result below, which is compared with the 11th order polynomial you show in the Instructions.pdf and that is coded in the Arduino software. Does this look reasonable? I ask because it is somewhat different from yours. My detector seems to be working fine, with reasonable rates and no other issues. 

Thank you,
Seth

[image]

spenceraxani, 2020-11-24:
Hi Seth,

That’s looking pretty good if you ask me. The difference is likely due to the shape of the waveform that you inserted to do the calibration. You can get an eye for it by simply looking at the signal on an oscilloscope — when you see a 50mV pulse on the scope, does the detector readout 50mV?

I didn’t have a very good waveform generator, so it could even be that your calibration is better than mine.

Thanks,

Spencer

claudepi, 2020-12-01:
Hi Spencer, 

Thanks for that suggestion. Indeed I do not get a 1-1 correspondence.

I plugged the detector into my scope via the BNC connection to measure the raw pulse from the SiPM. I made 15 measurements where I recorded the raw pulse amplitude on the scope and the corresponding pulse amplitude in the data readout (with my calibration coefficients used in the Arduino software to convert ADC to pulse amplitude):

[image]

These 15 values are plotted in the figure below in red with a best fit line. As you can see they do not lie on the 1-1 line. I also plotted (in green) what would be expected based on your calibration, i.e., using the ADC values from the readout and mapping those into a pulse amplitude using the 11th order fit from SDcard.ino. You can see that these are much closer to the 1-1 line (at least below ~300 mV), suggesting that something went awry in my calibration procedure. Back to the drawing board ....

[image]

Thanks,
Seth

claudepi, 2021-01-09:
Hello again,

After quite a bit of trial and error, I have figured out that there was indeed an issue with how I was scaling the reference pulse up and down with my arbitrary waveform generator. Now that I have fixed my procedure I believe I have a much better result (cal_run_final2), but it still looks a bit different from yours:

[image]

But when I do the same test as in my previous post (measure the amplitude of the reference pulse on my scope and compare it with the readout), I get a much better result with my new calibration (cal_run_final2):

[image]

So I think I am doing things correctly now. Though I am still curious/concerned that my result is different from your result. You mentioned above that it could be due to shape of the waveform that I inserted to do the calibration. Let me ask you a couple of questions in that regard:

(1) Did you use the same reference waveform for the entire range, scaling it up and down with your function generator to the desired input pulse amplitude?

(2) Did you have to periodically change the signal and reset thresholds to higher and lower values? I found that I had to change these (use lower thresholds at the lower end, and higher thresholds at the higher end) otherwise I wouldn’t get any counts.

I ask because in my calibration run above (cal_run_final2), I used a bunch of different reference waveforms. So I went back a did the same procedure, but using a single raw waveform for the entire range. And I did that for 3 separate reference waveforms (80 mV, 336 mV, 1320 mV) and compared the results:

[image]

There are some subtle differences, but they all lie reasonably close to one another. So I guess it doesn’t much matter, but I was just curious what you did.

Thanks again,
Seth

claudepi, 2021-05-23:
I wanted to provide an update here. Since my last post I built two new detectors because my other two died as noted in #63. I've done an electronics calibration for each of the two new detectors. They are slightly different from one another but similar to the ones I obtained above on the old (now dead) detectors. Here they are compared with the electronics calibration from Instructions.pdf:

[image]

(Note: my two new detectors are labeled "201" and "202"). 

To see the effect that these different calibrations have on the ADC and pulse amplitude spectra, I took ~24-hour runs of data on consecutive days. This figure compares data from my detectors when in coincidence mode where they both have the default calibration from Instructions.pdf:

[image]

The profiles shown are for the coincident secondary ("scnd") detector in the arrangement (the data were obtained on two consecutive days). You can see there is an offset between the two spectra when plotted versus ADC. Since both detectors use the same calibration, the spectra do not line up when plotted versus pulse amplitude. 

However, when I make a similar comparison, but here using the electronics calibration that I have derived separately for each detector (i.e., the first figure in this post), the peaks line up much better when plotted versus pulse amplitude: 

[image]

Perhaps this suggests that my calibration procedure and separate calibrations for each detector are correct/on the right track?

I do note that there are some small differences in the two spectra, particularly near ~40-60 mV. You noted somewhere (either here in the github or in one of the reference documents) that the shape of this peak is determined by how well the SiPM is coupled to the scintillator. So perhaps that coupling is different in the two detectors and explains the discrepancies between ~40-60 mV? I'd love to hear your thoughts about that. 

I'll leave my calibration fit coefficients here in case they are useful to anyone:

"201 detector"
{-7.861644222315437e-27,  4.306401989118021e-23,
 -1.018937662726167e-19,  1.363249919743100e-16,
 -1.134365246184855e-13,  6.088163496534007e-11,
 -2.117955692271122e-08,  4.685141882190339e-06,
 -6.291123433458015e-04,  4.714795861579866e-02,
 -1.525151750121887e+00,  3.656490996599823e+01};
 
"202 detector"
{-6.098239192418842e-27,  3.491651162355342e-23,
 -8.636983165340803e-20,  1.208625158349717e-16,
 -1.052847733764935e-13,  5.924823493423876e-11,
 -2.166636100240393e-08,  5.058223765355062e-06,
 -7.214146629642136e-04,  5.796146568638363e-02,
 -2.102406534552599e+00,  4.574866345492077e+01};

spenceraxani, 2021-06-01:
Hi Seth, 

Thanks for posting this rather large amount of work. I’m surprised by how much different our calibrations differ. I don’t have a great explanation for it. The best I can come up with is that we are surely using different waveforms to perform the calibration. 

Looks like you are on the right track though. It’s interesting that after using separate calibrations for the two detectors, your MIP peaks appear to line up better. This may indicate that shifts observed in terms of ADC counts may be partially attributed to the electronics rather than efficiency differences (optical coupling, scintillator light yield, ect.). I’ll have to look into this more for the next set of detectors. 

Thanks,

Spencer

claudepi, 2021-06-01:
Super thanks Spencer. I will close this and reopen if I find anything new. 
--Seth

closed 2021-06-01 by claudepi