spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #63: Arduino Failed

claudepi, 2021-02-10:
Hi Spencer,

Have you seen (or heard) about any issues with Arduinos failing when using pyserial?  Recently, I was evaluating how to make coincidence measurements with two detectors both recording data to the computer through the python script import_data.py. When running the script, I was having trouble getting both detectors to reboot within the 2 second window required to set them in coincidence mode. Sometimes they would start in coincidence mode and other times they wouldn’t. So I was playing around with trying to use software to reset both detectors/Arduinos once the serial connections were established, specifically by toggling DTR with setDTR in pyserial. At some point during this testing, one of my detectors stopped being recognized by the computer and would not power on. 

When I opened it up, I observed that the POW LED on the Arduino was dimmer than usual and that none of the other 3 LEDs would not light at all (Tx, Rx, and L). I removed the SiPM+scintillator and tested the voltage going to it and got a very small value, ~1 V or less. I then tested between the 5V pin and the GND pin on the Arduino and got ~1.7 V, and I got the same result when testing the 3.3V pin. So it looks like there is some problem with the USB power on the Arduino. I was able to confirm this by successfully powering the Arduino on the 5V and GND pins from an external power supply. When powered this way, everything seemed to be working fine and I was able to measure 29.5V going to the SiPM+scintillator. 

I googled a bit and found someone that had similar symptoms and they traced it to the Schotky diode (MBR0520LT1G) that is shown on the Arduino Nano schematic as the +5V Auto Selector. Unfortunately, that component is on the bottom of the Arduino and inaccessible, so it will not be possible to replace it without removing the Arduino from the main PCB. I guess it could be one of the other components on the bottom as well. As removing the Arduino from the main PCB seems essentially impossible, it looks like this detector is now bricked. 

I am curious if you have ever seen or heard about behavior like this?  Once I build a new detector, I would like to find a way to reliably ensure that the detectors enter coincidence mode when establishing the serial connections in import_data.py (i.e., without having to manually press the reset buttons).  Any advice you have on this front would be greatly appreciated, as I am keen on being able to record coincidence data to the computer via the python script. And I'd like to not fry my next detector, so I am nervous about using the same approach (setDTR in pyserial).  

Thanks,
Seth

spenceraxani, 2021-02-16:
Hi Seth,

I haven’t run into this issue before, and sadly do not have any information on the setDTR. 

I think your troubleshooting indicates a short somewhere. If it was drawing too much current for some reason, I could see it not properly being able to go into coincidence mode, and I could also see it not turning on or being recognized by the computer. It would be strange to all-of-a-sudden develop a short though. Your USB port can only supply so much power, so maybe that is why your external power supply is working — it’s able to supply sufficient power for the short as well as the detector. When plugged in, you can touch the components on the detector to see if anything gets hot. If the short is in the diode under the Arduino, replacing it is going to be difficult. I would start by looking at the Arduino circuit to see if you can just add a new diode to the top of the board.

If you are going to replace the board, I would take a new PCB and simple transfer the components from the other board to it. This takes ~1hour. It’s simpler then trying to remove the Arduino, which is possible if you have a second person and a hot air gun, but may cause some damage to the board.

One way to test for a short is to use a USB power meter. It will tell you the voltage and amperage drawn by the detector. Typically, these detectors take ~0.27W to run (which is actually often too small of a power to measure, so I measure the power draw from two of them—0.54W, or 5V @0.1amps). 

The detectors should almost always into coincidence mode if they are booted up at the same time. The following lines set up the coincidence:
  pinMode(3, OUTPUT);
  pinMode(6, INPUT);
  if (digitalRead(6) == HIGH) {
      SLAVE = 1;
      MASTER = 0;
      digitalWrite(3,HIGH);
      delay(1000);}

  else{
      delay(10);
      MASTER = 1;
      SLAVE = 0;
      pinMode(6, OUTPUT);
      digitalWrite(6, HIGH);}

If you removed the delay(10); line, it would probably always go into coincidence mode if they are booted up at the same time. You could also instead hardcode one of the detectors to MASTER =1; and the other to SLAVE = 1; and then the SLAVE would always be looking for coincidence signals no matter what.

Thanks,

Spencer

PaulHearn92, 2021-02-17:
Forgive me as this is off topic.
I'm a retired engineer in the UK and have a Radio Astronomy lab.  I'm looking at adding muon detection to my data flow.   It it possible to purchase a CosmicWatch-Desktop-Muon-Detector-v2 or a full kit of parts.  Again apologies for the interruption.
Paul.      [paul@hearn.org.uk]

claudepi, 2021-02-28:
Hi Spencer,

Thanks for all of the tips. A couple points of clarification on my original post:

- I was not having any issues with getting the two detectors to go into coincidence mode, in general. Specifically, I was able to place them in coincidence mode without any issues by manually pressing the reset buttons on the back of the detectors. What I was trying to troubleshoot was getting them to go into coincidence mode when running import_data.py. That was hit or miss, and I think it was related to the time it takes to loop through the various commands when setting up the serial ports. So that's what led me down the path of toggling the DTR (using pyserial commands) to reset the two detectors within the 2000 ms window using software commands (rather than manually pressing the reset buttons). 
- I should have been more clear about my external power supply. It's not anything fancy, just a breadboard power supply that I use that I know supplies ~5V. Also, I misspoke in my original post: I successfully powered the detector and observed proper functionality with this external power supply when powering directly via the Vcc and GND pins on the arduino (not the "5V and GND pins" as I originally stated). 

Since my original post I have also learned that the second detector involved in the above coincidence testing is suffering from similar issues. I did not notice this at first because I was focused on troubleshooting the first detector.  The issues are nearly identical to what's in my first post. 

To answer a couple of your follow up questions:  

- When powered, nothing on the top (accessible) part of the arduino seems to be getting hot. Ditto for the main/SDcard PCBs.  
- Measuring the power draw with a USB power meter I find that one detector draws ~0.20 W (0.04A and 5.04V). When I power it with my breadboard power supply directly to Vcc and GND pins on the arduino (i.e., not through the mini-USB) I get essentially the same, ~0.19 W (0.04A and 4.95V). So I'm not sure that powering via the external supply is giving it more power that might help with a short. 

Who knows what's going on and without knowing exactly what component is bad or where a potential short is, adding a new diode to the top of the arduino will be difficult (plus that’s probably beyond my current skill set). Thanks for the tips about potentially replacing the arduino and/or transferring the components to a new PCB but desoldering is also not my best skill.
 
It’s funny because on the very first detector that I built about 4 months ago, the arduino began to smoke immediately when I first powered it on (I am currently 0 for 3 on builds!). At that time I thought I must have soldered something incorrectly but now all of this makes me think that maybe I got a bad batch of arduinos. I’ve gone back and read a bunch of 1-star reviews for the Chinese clone arduinos on Amazon and have seen enough reports of similar things to make me nervous about these cheap clones, so I think I am going to suck it up and buy the real arduinos and start over.

Thanks again!
--Seth

spenceraxani, 2021-03-01:
Hi Seth,

I’ve probably purchased ~300 Nanos at this point and I should say that one set of x30 I bought through Ebay didn’t work. I don’t recall the problem with them, although I know they sit in my lab with a “broken” piece of tape attached to them.

If you have a spare Nano, perhaps you can play with it a bit while it is not soldered to the CW main PCB.

The import_data.py script does have some delay associated with setting up each detector. Are you plugging both detectors into the computer independently (two usb ports)? There is a line in the import_data.py, which waits for 1 second between setting up the independent detectors:
	
	line 219: time.sleep(1)

You could try decreasing this and seeing if that solves your problem. If you are not plugging in both detectors (the coincidence connection provides power to a second detector), then I don’t see this causing the issue. 

Another option you could try, is to set up one of your detectors as the COINCIDENCE detector by hard coding it into the .ino file. This way, it would only ever record an event if it saw a coincidence pulse come in through the coincidence connection. You would then just have to remember which detector is your coincidence detector.

Hope this provides some help.

Thanks,

Spencer

claudepi, 2021-03-07:
Spencer,

Yes, I was plugging both detectors into the computer independently with two USB ports. Once I get my new detectors built, I will trying decreasing the sleep time to see if that helps.
 
And thanks for the tips about modifying the .ino code to hardcode one of the detectors as the coincident one.  That's probably going to be my best option going forward.
 
I'm going to close the issue - if I find out any new information I will add it here later. 
 
Thanks again,
Seth

closed 2021-03-07 by claudepi