spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #64: SiPM Module seemly not working

wvols, 2021-04-15:
Hi,

I have built 3 detectors thus far and none of them seem to work. They all are giving off ~29.6 V but none are not giving off any counts at any normal rate.
90% of the time there are zero counts happening, sometimes when the detector is held and fiddled with it counts momentarily at an erratic rate (anywhere from 1-3 a second up to 10+ a second) then will stop again unexpectedly. 

The boards have been reflowed/resoldered and checked the connections of all the parts with another much more experienced person to confirm I am not just inherently bad at it. 

So, the thinking is there is something messed up in the SiPM module. 
The resistivity test was done on all modules getting some resistivity one way and infinity the other way.
    How sensitive are the SiPMs when soldering? Like if you are heating the board/solder/SiPM for more than a couple of seconds 
       will the device bust easily?
    There seems to be no difference/nothing discernable at all when analyzing the pulses on an oscilloscope with it mostly being noise at all test points. Using covered with scintillator and uncovered with no scintillator.

Jumper cables were used to try and isolate a connection problem and it had no effect. 
Sometimes pulses are counted even with the SiPM module disconnected and running a finger over the SiPM connector.

Is it okay to remove/replace the SiPM when the base circuit board is turned on? Different combinations of SiPM modules and baseboards have been tried. 

Two of the three detectors when booted up will give off a single count then nothing from there. 
On a separate note, another (maybe uncorrelated?) problem is one detector has a slow refresh rate and will skip a second on the OLED about every other second.

Has anyone else had this problem?
What would you suggest I try to do to fix it? Is there another way to isolate the SiPM to confirm it is working? 

Thanks for creating this project it has captured my interest and I am eager to get it working.

spenceraxani, 2021-04-15:
Hi,

Thanks for the detailed description of your problem. Since the DC-DC booster appears to be working, and you do not see a signal out of the BNC connection (you mentioned the test points, but I’m assuming you also tried the BNC directly), the issue is likely with the SiPM+scintillator module. 

If this is the case, then the most likely cause is a poor connection on the SiPM, or the SiPM is orientated incorrectly. The module itself is rather robust, and I wouldn’t worry about overheating it. Since you mentioned that you are worried about the heat, my bet is that you ended up with a cold solder joint on the SiPM that disconnected when interfacing it with the scintillator. I would start there, along with double checking the solder connections on the resistors and capacitors on the SiPM PCB board. The erratic counting, I would say, is also an indication that you have a poor connection on the SiPM.

The other thing to double check is that the orientation of the SiPM is correct. Can you double check that the SiPM Pin1 is correctly positioned with pad 1 on the board?

I wouldn’t worry about the rest of the board until you see pulses out of the BNC connection. But if you see pulses, and the boards still don’t work, feel free to send me pictures of the boards. 

Regarding the slow refresh rate on one of the detectors; I have seen this before on one of the more recent detectors I built, although I didn’t dig into it to figure out if it was related to the OLED or the Arduino. I found that it was only noticeable when the detector booted up and didn’t affect the performance. Could try changing the OLED screen. I set the refresh rate on all detectors to 1Hz anyways, while running. 

I would turn off the detector before switching SiPM modules. 

Thanks,

Spencer

RebekahSebok, 2022-03-26:
Hi Spencer,

We are having the same issues as described by wvols. We have so far built two of the SiPM boards and neither are registering any counts. We have the correct voltage at the 6-pin connection and the sensor passes the resistance test shown in your video, but the BNC connection doesn't show any pulses. We have reflowed the solder connections and we are sure that pin 1 is placed correctly for the SiPM. Do you have any suggestions for narrowing down where our issue may be? 

Thank you,

Rebekah

spenceraxani, 2022-03-27:
Hi Rebekah,

Since you have the correct voltage being supplied to the SiPM, when it is interfaced with scintillator and made light tight, you should see pulses out of the BNC. Assuming you setup your oscilloscope correctly, the possible causes are:

1. The SiPM is not correctly mounted. This is likely your issue since you mentioned that the sensor does not pass the resistance tests. Note that the resistance tests requires some light. Not passing this test either means that your SiPM is either rotated (recheck the pin orientation), or that the solder connection is not adequate. Since you mentioned that you think you have the correct orientation, the issue is likely that you need to improve the solder connection.

2. There is a light leak. If you have a large amount of photons entering the scintilator because of a light leak, then the SiPM will not register individual pulses. 

3. Your BNC is not soldered. Simply check the solder connection on the BNC connector.

4. The resistors on the SiPM PCB going to pin 3 are not mounted, or disconnected. This would be fixed by improving the solder connection.

Thanks,

Spencer

RebekahSebok, 2022-03-30:
Hi Spencer,

Thank you for your quick reply! We touched up all of the connections on the SiPM board and went through with a multimeter to make sure there were no shorts and then we re-wrapped the whole SiPM assembly and put it in a small cardboard box. We realized that only one of our main boards worked, but combined with the re-done SiPM assembly we have a fully functioning detector.

Thanks for your advice,

Rebekah