spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #65: recording master and slave data to computer

mantonacci, 2021-06-02:
Dear Spencer,

Thank you for creating and maintaining this project, it has been a wonderful and affordable way to get our undergrad students involved in nuclear physics experiments! 

We (my students and I) were wondering if there was Arduino code to have both the master and the slave data recorded to a computer?

Thank you!

spenceraxani, 2021-06-02:
Hi Michael,

Thanks for the email.

I tend to save all data to the microSD card, which indicates in the file name which detector was setup in Master versus Coincident configuration. 

If you would like to record both detectors to a single computer, you can simply plug them both in via USB, then when prompted to “select Arduino port” simply select both ports by separating them with a comma. This will record the data from both detectors to a single file, and each event will be labelled with the appropriate detector name. 

The problem with this, is that both detectors will be automatically put into Master mode. You could manually select the coincident events in the file by looking at the time stamps from the computer, but if this is cumbersome, then the other option would be to hardcode one of your detectors into Coincident mode. To do this, simply force the SLAVE variable to be 1, and MASTER to be 0 in lines 76 through 87 in the Arduino OLED code (79-93 in SDCard.ino).

If you would like two separate files for the two detectors, you can run the import_data.py program twice, selecting each detector independently. 

Hope this answers your question.

Thanks,

Spencer

mantonacci, 2021-06-02:
Hi Spencer,

Thank you so much, that is exactly what I was looking for. 

Have a safe and productive summer!

Michael

closed 2021-06-02 by mantonacci