spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #67: OLED Screen no longer counting and LED not flashing

Draconum, 2021-07-12:
Hi Spencer,

We had a functioning Cosmicwatch detector built a year ago and the LED light and OLED screen are no longer registering muons with energies above the threshold as recorded on an oscilloscope. With the test points, TP1 seems to be similar to the Raw SiPM pulse, however, TP2 just exhibits noise and TP3 shows a continuous pulse of 10mV. We have made sure that the legs of the op amp and the DC-DC booster are well soldered and have gone through the rest of the troubleshooting. We currently think that the issue is with the amplifying circuit and wondered what else could be causing this.

Thank you!

Draconum, 2021-08-10:
The voltage on the DC-DC Booster appears as negligible or 0.4V even though we have double-checked all connections on op-amp and the signal on the oscillope is shown below:
[image]

spenceraxani, 2021-08-11:
Hi,

If you aren’t able to see ~29.6V between the HV and GND pins on the 6-pin header, then this problem lies in the part of the circuit labelled DC-DC booster. I have run into situations where I’ve needed to replace the IC (LT3461) as well as the inductor. But, 90% of the time, the problem is with a connection problem on one of the legs of the IC. The op-amp (LT1801) is completely independent of this part of the electronics. The DC-DC booster problem needs to be solved before troubleshooting anything downstream.

If you do indeed not measure the correct voltage on the HV pin, then no signals will be present from the SiPM (BNC and TP1). Once you get the booster solved, feel free to let me know if you are still running into problems with TP2.

Thanks,

Spencer

closed 2021-08-24 by Draconum