spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #68: [REQUEST] PCB CAD sources

Blaster1920, 2021-09-01:
Hi,
I'm very interested in replicating this project, is there any chance you could upload the source files of the PCB and Schematics?
I'd like to apport some modifications and it would make it much faster.
Best

PaulHearn92, 2021-09-02:
Thanks Blaster1920, I too have the same questions. And..
Do we know how many CosmicWatch detectors are operating, And..
Is there a project to capture all the data, And...
Can a kit of parts be purchased for v2.

Blaster1920, 2021-09-02:
Interesting questions.
I would have the ability to produce several PCB kits, if I can modify the
PCB to use, perhaps, an esp32 which can be easily connected to the wifi and
synced over.

spenceraxani, 2021-09-02:
Hi Paul,

My guess is that there are approximately 1,000 detectors in operation. I’ve built ~300-400 myself, and then have probably spoken to several hundred (maybe thousand) people building them. 

Eventually, the plan is to release a new version of the detector, with an updated website that logs the location and rate. The pandemic has slowed this down. Normally, I would bring in a few undergraduates to supply kits to younger students or people in nations that cannot afford the detector components. But, in general, kits are not available for purchase.

Thanks,

Spencer

spenceraxani, 2021-09-02:
Hi Jacopo,

Would you be able to contact me through my email ***@***.***), to let me know what you would like to do?

I’m hesitant to provide the source files for the PCB design. Although I understand why some people may be interested in updating the microcontroller (I have other version that run on the ESP32, ESP8266, Teensy 3.6, and Teensy 4.0), the issue lies with project maintenance. Providing technical support to people building detectors is actually rather time consuming (several hours per day on average). Introducing another version (or modifications) complicates this. I’ve run into this issue many times with the previous versions of the detectors.

Thanks,

Spencer

Blaster1920, 2021-09-02:
I sent you an email on the public address you've got on your github.