spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #71: SiMP PCB Reflective surface

Airbasti, 2021-12-15:
Hello, 
i am currently in the process of building two of the detectors. The PCBs just arrived and i have noticed that the SiPM PCB does not have a reflective surface around the point where the SiPM is placed. Is this a failure by JLC PCB or did i send the wrong code or will the reflective surface be added by me later?
Thank you in advance
Sebastian

P.S
Does anyone have experience with homemade optical gel? It seems to be out of stock everywhere at the moment

spenceraxani, 2021-12-15:
Hi,

It looks like the PCB Gerber files for the ‘PCB_Gerber_files_wLargerSnapOffConnectors.zip’ were missing the reflective plane. The detector will still work fine without it. The scintillator is surrounded by an aluminium foil to help with reflection. Not having the reflective plane probably means that you lose a few percent of photons — which is insignificant.

Thanks,

Spencer

Airbasti, 2021-12-16:
Hello, 
thank you for the quick advice. Do you think surrounding the SiPM by aluminium foil (hence just making the left out square of the scinti wrapping smaller) will result in similar results or will this cause other problems? I really want to as good results as possible.

PaulHearn92, 2021-12-16:
In answer to the optical gel question, I think Silicon grease will work.  It has a refractive index of 1.465.  We at UKRAA are testing this and the ‘no grease’ option.  So far we have built 6 all with optical gel, our interface tests will be done on the next batch of 10, however there looks like a delay in the SiPM’s. I would be interested in Spencer’s take on this.

 

Paul Hearn (UKRAA)

spenceraxani, 2021-12-26:
Hi Paul,

Apologies for the late reply. The optical contact should not have any noticeable timing effect on the signal from the SiPM. The only conceivable delay that I could imagine, would be the result of an extra reflection off the interface surface, which would only amount to <<100ps delay in a fraction of the photons. This is much shorter than the rise time of the SiPM (~1ns), so it wouldn’t be possible to extract this delay. Perhaps something else is going on? How are you observing a delay?

The optical grease, however, does have a large impact on the efficiency of the detector to collect photons. Without it, there is an air gap (n = 1). The change in the refractive index will cause some photons that would have been absorbed by the SiPM to be reflected back in to the bulk material of the scintillator. I typically use optical gel, but I have been told by one of my colleagues that vaseline or petroleum jelly has a similar index of refraction (n = 1.45 - 1.48) to scintillator and can work well. 

Thanks,

Spencer

PaulHearn92, 2021-12-27:
Hi Spencer,

 

Your answer is far more interesting than my question, many thanks for the information.

 

My comment regarding delay was referring to the delivery time, we have been quoted about 10 weeks.

 

Best wishes and have a great 2022,

 

Paul.

 

Paul Hearn

 

¦¦¦  British Astronomical Association  ¦¦  Radio Astronomy Section  ¦¦

Burlington house, Piccadilly, W1J 0DU

 <https://britastro.org/section_front/24> https://britastro.org/section_front/24

 <https://www.youtube.com/user/britishastronomical/playlists> https://www.youtube.com/user/britishastronomical/playlists

   

e      ***@***.***> ***@***.***

a     RG6 1BU UK

t     +44 (0)7967 388 578

spenceraxani, 2021-12-27:
Hi Paul,

Aha, I see, my mistake. There was an expected delivery date of Dec. 15th 2021, although it now looks like it’s been pushed back to Feb. 11th 2022. Hopefully they aren’t delayed again. We’re also waiting on SiPMs for various projects. 

Thanks,

Spencer