spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #72: Existing coincidence data sets for exercices

mtetrault, 2022-01-04:
Hi,

I'm preparing basic training exercices/homework on simple data analysis where we will eventually have CosmicWatch units assembled and generating the data to process. In the meantime I'm hoping to use public or loaned data sets from coincidence experiments, for example to reproduce figure 13 in the reference paper (https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/blob/master/ThePhysicsPaper.pdf).  jgOhYeah was kind enough to share some files, however I'm having issues associating the coincidences post-acquisition based on the timestamps. After discussions, I assume my issue there is that the clocks between the two arduinos drift apart, and so far I'm unsucessful in finding a workaround. 

The paper mentions a customized Arduino code to directly tag coincidence/non-coincidence data. Based on figure 13, am I correct to assume the tagging is back-annotated to the master data? Is the raw data for this figure available somewhere in CSV form?

Thanks!

spenceraxani, 2022-01-04:
Hi,

You bet, here is the data:

The one thing to keep in mind, is that this file has an extra column (the final column) that labels coincident events with a ‘1’ and non-coincident events with a ‘0’. That is, the default Arduino code was modified to spit out this extra bit of information. Normally, two detectors are connected in coincidence, the coincidence detector will create a file with the prefix label ’S’ indicating that it is only saving the coincidence detections. 

You are correct in that the clocks on the two detectors will drift apart over time. If two detectors are connected via the coincidence cable, coincident events are tagged through a digital signal, rather than the time stamp. However, if the data was taken using the import_data.py python script via a computer, the computer will add a global time stamp to each event that is accurate to within ~5ms. 

Feel free to let me know if this does not answer your question fully.

Thanks,

Spencer

> Hi,
>
> I'm preparing basic training exercices/homework on simple data analysis where we will eventually have CosmicWatch units assembled and generating the data to process. In the meantime I'm hoping [...]

##########################################################################################
### CosmicWatch: The Desktop Muon Detector
### Questions? ***@***.***
### Comp_date Comp_time Event Ardn_time[ms] ADC[0-1023] SiPM[mV] Deadtime[ms] Temp[C] Name
##########################################################################################
Device ID: Billy
0 655 492 134.92 0 19.61 0
1 2064 359 69.86 113 19.61 1
2 2301 307 54.48 227 19.61 1
2 3719 67 18.51 348 19.61 0
2 5283 435 104.92 463 19.61 0
2 5749 56 16.76 590 19.39 0
2 7788 176 30.09 710 19.61 0
2 7964 71 19.06 827 19.61 0
3 8558 271 46.93 946 19.61 1
3 11081 204 34.73 1069 19.61 0
4 11486 271 46.93 1191 19.61 1
4 12879 71 19.06 1312 19.61 0
4 15146 56 16.76 1432 19.61 0
4 16548 216 36.85 1556 19.61 0
4 17253 179 30.55 1677 19.61 0
4 18463 96 21.67 1797 19.61 0
5 18912 643 246.01 1920 19.61 1
5 19259 391 82.35 2040 19.61 0
5 21594 88 20.96 2159 19.61 0
6 22176 350 66.46 2278 19.61 1
7 23206 295 51.75 2400 19.61 1
7 23346 71 19.06 2520 19.61 0
7 23495 199 33.86 2644 19.61 0
7 25125 103 22.23 2767 19.61 0
8 25507 270 46.71 2887 19.61 1
8 25728 184 31.36 3005 19.61 0
9 27982 419 97.20 3127 19.61 1
10 28710 364 71.71 3244 19.61 1
11 30545 275 47.73 3360 19.61 1
11 30851 49 15.48 3479 19.61 0
11 31839 327 59.28 3598 19.61 0
11 32067 176 30.09 3716 19.61 0
12 32253 323 58.48 3834 19.61 1
13 33536 192 32.67 3956 19.18 1
13 37197 88 20.96 4073 18.32 0
13 37751 433 104.37 4194 18.00 0
13 37899 159 27.66 4314 18.32 0
14 38774 319 57.64 4436 18.32 1
15 39677 409 91.12 4555 18.00 1
16 40345 248 42.69 4676 18.32 1
17 41352 263 45.37 4801 18.00 1
17 44365 158 27.53 4920 18.32 0
17 44748 231 39.55 5040 18.96 0
18 45918 204 34.73 5160 19.18 1
18 46224 286 49.78 5283 19.18 0
18 47855 51 15.86 5402 19.18 0
[... 1683 lines cut ...]
606 2175383 304 53.73 208893 18.96 1
606 2177185 57 16.93 209010 19.18 0
606 2178171 65 18.22 209126 18.64 0
606 2178975 60 17.43 209244 19.18 0
607 2179117 323 58.48 209362 19.18 1
607 2181522 224 38.29 209478 19.18 0
607 2185416 211 35.96 209597 19.18 0
607 2185989 65 18.22 209713 19.18 0
607 2187433 142 25.64 209829 19.18 0
607 2188801 312 55.68 209948 18.96 0
608 2189069 270 46.71 210070 18.96 1
609 2191245 359 69.86 210193 19.18 1
609 2191374 131 24.54 210311 18.64 0
609 2191800 108 22.63 210433 19.18 0
609 2192405 179 30.55 210552 18.96 0
609 2192987 97 21.75 210670 18.64 0
610 2195110 419 97.20 210791 19.39 1
610 2195276 49 15.48 210909 18.96 0
610 2195497 79 20.03 211027 19.18 0
611 2196743 455 117.03 211144 18.96 1
611 2200163 60 17.43 211264 19.18 0
611 2201995 193 32.84 211382 18.96 0
612 2203418 472 125.87 211500 18.96 1
613 2204021 348 65.64 211622 19.18 1
613 2204530 7

mtetrault, 2022-01-04:
Hi Spencer,

Thanks for the quick reply! I wrote a quick script to make the histograms and it comes out wonderfully.

Just to make sure, is the data complete or did the text box inadvertently cut off due to the number of characters? While I have plenty of detected events with what's available in the post for training, if there are more events it will likely make  a smoother histogram. 

Cheers

spenceraxani, 2022-01-04:
Hi,

Can you please clarify what you mean by "text box”? The data I sent you should be the exact data that I used to make Fig. 13.

In this case, more event won’t likely help too much more. The Arduino Nano ADC is pretty terrible. I have Fig. 13 binned such that it looks OK. Once you rebin it, you start seeing some strange structure related to the electronics and ADC. This structure doesn’t go away with more statistics. 

Thanks,

Spencer

mtetrault, 2022-01-04:
Hi Spencer,

On first look, the last line of data looks incomplete, but since this is the data for the figure, then I'm guessing that's simply where the file writing was cut off at the end of the experiment. I've removed the last (incomplete) line from my working file to not worry about this corner case.

The "text box" referred to the message box of the issue tracker. I had though it might have a length limit.

Cheers

closed 2022-01-04 by mtetrault

spenceraxani, 2022-01-04:
I see. I’m actually just using email rather than the issue tracker. I wonder if that has something to do with it. You may be able to simply download the file directly to your computer rather than opening it in the git text handler. The file looks fine to me; the last event should be 12906.

mtetrault, 2022-01-04:
Ah, then yes, very likely due to email. Unfortunately the data appears as plain text in both email and in the issue tracker, and stops at coincidence #613. There is no file attachment to download.

Can I trouble you to attach the file in the issue as an attachment? Not sure if it automatically detects and converts plain text files...

Thanks

spenceraxani, 2022-01-04:
Ah, I see the issue now. Let's see if this works.

[S.TXT](https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/files/7809589/S.TXT)