spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #73: 1 count and then no detection

halifaxjack, 2022-01-05:
Hi Spencer, 

I'm new to electronics and am learning a lot as I go through this project, so thank you for making this.

I've assembled 2 detectors and am getting the same 1 count error on both of them. I'm getting the correct voltages according to the instructions, and my SiPM is giving resistance. I've played around with your code and changed the thresholds but still get the 1 count error.

Here are some pictures of my hardware, incase it's an assembly issue.
[image]
[image]
[image]
[image]
[image]

Cheers,
And thanks again!

Jack

spenceraxani, 2022-01-05:
Hi Jack,

Thanks for the nice images. One thing that pops out is the lack of tape or tape residual on your SiPM PCB. I’d like to first confirm that you are wrapping the whole SiPM PCB assembly in black tape before booting it up. It will have to be light tight to work.

Your solder joints look well done, although there is likely a cold solder or poor connection somewhere. This tends to be the most likely issue, so I would start there — hit every solder joint with some new solder to ensure a good connection. For example, it looks like one of the legs on the op amp may not be connected (image below). R3 looks OK, but could also be suspect.

If the above two things do not solve your problem, can you remove the temperature sensor? Twice in the past month-or-so I saw people with faulty sensors.

Finally, if you are changing the thresholds, I would increase them. Maybe bring them both up to ~60 ADC counts.

Thanks,

Spencer

halifaxjack, 2022-01-05:
Hi Spencer, 

Thanks! I hit a few joints with some extra solder and light tightened by SiPM PCB and its working now! Now I need to increase that threshold because I'm getting some crazy counts.

Cheers 
Jack

closed 2022-01-05 by halifaxjack

spenceraxani, 2022-01-05:
Glad to hear it. I would keep the threshold at the default value. Perhaps you have a light leak? You really need it to be light-tight. The count rate should be single Hz level.