spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #74: SiPM Optical coupling

PaulHearn92, 2022-01-06:
I did a test between PM Optical coupling gel and Petroleum Jelly.   I could see no difference in count rate.  Is Optical coupling gel Petroleum Jelly or something else.  We are considering what to include with our Cosmic Watch kits.
Paul Hearn.

spenceraxani, 2022-01-06:
Hi Paul,

I’m not sure of the chemical composition of optical gel, although I would bet it’s different than petroleum jelly. While their refractive indices are similar, the optical absorption is different. Optical gel is designed to have the correct index of refraction, a high transmisivity to the wavelength of electromagnetic radiation that PMTs are most sensitivity to (~420nm), and a specific viscosity that helps remove air bubbles and provide a long term interface. With petroleum jelly, it’s purely a coincidence and you likely lose extra photons. 

Thanks,

Spencer

PaulHearn92, 2022-01-06:
Thanks Spencer,

 

One again a great and swift answer…

PaulHearn92, 2022-01-07:
Dear Spencer,

 

I believe the energy in mV from the SiPM is a function of the muon event energy and the path length in the scintillator.  Do we know the minimum muon path length to give a output signal.  I’m looking to see if the SiPM output can be related to the originating muon event.

 

Best wishes,

 

Paul.

 

Paul Hearn

 

¦¦¦  British Astronomical Association  ¦¦  Radio Astronomy Section  ¦¦

Burlington house, Piccadilly, W1J 0DU

 <https://britastro.org/section_front/24> https://britastro.org/section_front/24

 <https://www.youtube.com/user/britishastronomical/playlists> https://www.youtube.com/user/britishastronomical/playlists

   

e      ***@***.***> ***@***.***

a     RG6 1BU UK

t     +44 (0)7967 388 578

spenceraxani, 2022-01-07:
Hi Paul,

That’s correct. We typically assume cosmic-ray muons to be a minimum ionizing particle, which will deposit 2-3MeV per cm travelled in this scintillator (density of ~1g/cm^3), regardless of the muon energy. Since the scintillator is ~1cm thick, the majority of the muonic events will deposit 2-3MeV, which is larger than the energy deposited by most backgrounds. There are gamma rays that are above 1MeV (like 40K and 60Co), although since we are using low density scintillator, the gamma ray is more likely to Compton scatter, depositing a small fraction of energy in the scintillator and scatter out.  This is why the muons pop out at higher SiPM values and backgrounds populate the lower SiPM values, as shown in Fig. 13 of the PhysicsPaper.pdf:

https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/blob/master/ThePhysicsPaper.pdf <https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/blob/master/ThePhysicsPaper.pdf>

The energy resolution on the detector is extremely poor though. I would bet that the minimum path length for a muon to trigger the detector is somewhere around 2mm. 

Thanks,

Spencer

PaulHearn92, 2022-01-07:
Hi Spencer,

 

Thanks again for the great answer.  Has anyone ever tried to measure the energy resolution.

 

(todays job polishing ten scintillators for batch 3).

 

Best wishes,

 

Paul.

 

Paul Hearn

 

¦¦¦  British Astronomical Association  ¦¦  Radio Astronomy Section  ¦¦

Burlington house, Piccadilly, W1J 0DU

 <https://britastro.org/section_front/24> https://britastro.org/section_front/24

 <https://www.youtube.com/user/britishastronomical/playlists> https://www.youtube.com/user/britishastronomical/playlists

   

e      ***@***.***> ***@***.***

a     RG6 1BU UK

t     +44 (0)7967 388 578

 

https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/blob/master/ThePhysicsPaper.pdf <https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/blob/master/ThePhysicsPaper.pdf>

spenceraxani, 2022-01-07:
Hi Paul,

Yes, multiple people have tried, but you cannot extra any mono-energetic gamma-ray peaks since most interactions are Compton scatters. So, you’re pretty much left with the single datapoint from the minimum ionization from muons. The detector does not work for gamma-ray spectroscopy.

Thanks,

Spencer

tony1tf, 2022-01-15:
Regarding optical coupling, I have just used some silicon grease. This seems to work OK and appears to give plenty of sensitivity to muons. An old luminous watch causes the LED to light almost continually.
I see no difference in the spectra of cosmic rays or radium. I have been copying and pasting from the serial monitor of the Arduino app into an Excel spreadsheet - I have put all 4 ADC readings into the file to measure droop and pileup as well (rare) - in fact I see no reason to average the readings - the first one which is used to trigger the code is perfectly OK. I attach a typical spectrum.

Tony - BAA RAG 
[image]

spenceraxani, 2022-01-18:
Hi Tony,

A continually blinking LED most likely indicates either that the detector is not light-tight, or that you have a poor solder connection on the SiPM. 

Radium is an alpha emitter. Alpha particles cannot penetrate the black tape or aluminum foil. Slight differences can be observed using gamma rays, although the high probability of Compton scattering (rather than photoelectric absorption) due to the low density of the scintillator, and the small photocathode coverage, the detector cannot be used for gamma-ray spectroscopy. You will find that mouns do tend to populate the higher SiPM peak values though. 

Thanks,

Spencer

tony1tf, 2022-01-18:
> <spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/issues/74/1015344139@
> github.com>

HI Spencer

The detector works fine - I was just pointing out that I get plenty of
counts from a luminous watch. A WW2 luminous compass runs at about 150 cps.
B/G rate is around 1 - 2 CPS depending on which part of the UK I am in.

Tony

Airbasti, 2022-04-06:
> Regarding optical coupling, I have just used some silicon grease. This seems to work OK and appears to give plenty of sensitivity to muons. An old luminous watch causes the LED to light almost [...]

Hello Tony,

could you point out which Silicon grease you used? I am from germany and it's hard to find the right grease here. I am already overthinking the coupling but i would love to have just one example of a manufacturer that worked out.

Thank you in advance

PS: @PaulHearn92 same would be interesting for the petroleum jelly (is it completly transparent for?)

tony1tf, 2022-04-06:
> <spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/issues/74/1090481855@

Hi Airbasti

The silicon grease I used was a very old tube (probably that I have used on
plumbing fittings etc.) It is called 'Servisol' from Ambersil Ltd of
Basingstoke. The tube says 'High Electrical Insulating and Water Repellant
properties. Rubber-Plastic Lubricant, High Temperature Resistant.
I did a Google search and found it here:

https://cpc.farnell.com/servisol/200002000-50gm/silicone-grease-50g-tube-servisol/dp/SASILGRSETUBE

Tony

Airbasti, 2022-04-06:
> Hi Airbasti The silicon grease I used was a very old tube (probably that I have used on plumbing fittings etc.) It is called 'Servisol' from Ambersil Ltd of Basingstoke. The tube says 'High [...]

Perfect! Thank you so much! :)

PaulHearn92, 2022-04-07:
I built one detector with petroleum jelly and the result was the same as the one built with coupling gel.  I may have been lucky, the refractive index is the same, however the viscosity and melting point is very different.  In the long-term petroleum jelly may well deteriorate.

 

UKRAA are now selling kits and built detectors.  We decided to include with our kits optical coupling gel.  If you want a small syringe (0.1ml) enough for about three I can get a price.

 

Best wishes,

 

Paul. ***@***.*** <https://github.com/PaulHearn92> )

 

Paul Hearn

Radio Astronomy Section Director

 

¦¦¦  British Astronomical Association  ¦¦ 

Burlington house, Piccadilly, W1J 0DU

 <https://britastro.org/section_front/24> https://britastro.org/section_front/24

 <https://www.youtube.com/user/britishastronomical/playlists> https://www.youtube.com/user/britishastronomical/playlists

   

e      ***@***.***> ***@***.***

a     RG6 1BU UK

t     +44 (0)7967 388 578