spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #75: iPython Note Book

tony1tf, 2022-01-17:
Hi Spencer,  All I want to do is plot the data from my muon detector. I can do it in Excel, but that's rather clunky. The system you have set up seems to use iPython note books, which looks incredibly complicated, and is something I don't feel I want to learn. Can you point me to any simple python code which I can use on a MacBook to plot the output of my detector either in batches or in real time? I have managed to edit the code which receives data from the detector and saves it to a text file to run under Python3, since the libraries wouldn't install in Python. Lots of '()'s needed adding to print statements, and substitution of 'input' for 'raw_input'.
Tony

spenceraxani, 2022-01-18:
Hi Tony,

I don’t think I have anything simpler than that in the iPython script, although, like you mention, it was written for python 2, which means the print statements and raw_input may need to be update.

To use the iPython code in a python script (copy and paste the iPython cells), you should only have to remove the first three lines:

%matplotlib inline
from IPython.core.display import display, HTML
display(HTML("<style>.container { width:100% !important; }</style>”))

You will also have to replace the file name with your file name.

Thanks,

Spencer

tony1tf, 2022-01-31:
Hi Spencer
I am trying to run your example plotting script, and have put my file name into the code, but I can't seem to get round the following errors:

`TypeError`                                 Traceback (most recent call last)
/var/folders/43/v1d780lj3dvff3z8g6t_qphr0000gn/T/ipykernel_63770/2022836374.py in <module>
      3 #    2. Give the size of binning in time you would like
      4 cwd = os.getcwd()
----> 5 f1 = CWClass(cwd + "/background_31Jan2022.txt", bin_size = 10)
      6 # f2 = CWClass(cwd + "/Sheilding/bkg_nobox_sheild_long.txt", bin_size = 60)

/var/folders/43/v1d780lj3dvff3z8g6t_qphr0000gn/T/ipykernel_63770/2328829766.py in __init__(self, fname, bin_size)
     80             return sum(check == l)==len(l)
     81 
---> 82         if not sequential(event_number):
     83             print('Events in file are not sequential.')
     84 

/var/folders/43/v1d780lj3dvff3z8g6t_qphr0000gn/T/ipykernel_63770/2328829766.py in sequential(l)
     67         def sequential(l):
     68             l = np.asarray(l).astype(int)
---> 69             check = range(min(l),max(l)+1)
     70 
     71             if len(l)!=len(check):

TypeError: iteration over a 0-d array
`
Tony

spenceraxani, 2022-01-31:
> Here it as.
> Could you send me the file you are trying to read?
> <spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/issues/75/1025882067@
> github.com>
> b'##########################################################################################\r\n'b'### CosmicWatch: The Desktop Muon Detector\r\n'b'### Questions? ***@***.***\r\n'b'### Comp_date Comp_time Event Ardn_time[ms] ADC[0-1023] SiPM[mV] Deadtime[ms] Temp[C] Name\r\n'b'##########################################################################################\r\n'Device ID: b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\r\n'2022-01-31 14:15:29.843680 b'1 486 170 29.19 348 21.01\r\n'2022-01-31 14:15:29.943407 b'2 586 342 64.07 350 21.09\r\n'2022-01-31 14:15:30.039604 b'3 682 95 21.58 352 20.71\r\n'2022-01-31 14:15:30.304830 b'4 947 310 55.16 354 21.09\r\n'2022-01-31 14:15:30.655584 b'5 1297 369 73.42 444 20.90\r\n'2022-01-31 14:15:31.103117 b'6 1744 291 50.97 446 20.76\r\n'2022-01-31 14:15:31.447996 b'7 2000 55 16.58 448 20.74\r\n'2022-01-31 14:15:31.862813 b'8 2504 175 29.94 538 21.14\r\n'2022-01-31 14:15:35.187893 b'9 5827 53 16.22 804 20.79\r\n'2022-01-31 14:15:35.699327 b'10 6337 170 29.19 893 21.14\r\n'2022-01-31 14:15:35.979665 b'11 6616 438 107.18 895 21.01\r\n'2022-01-31 14:15:37.072930 b'12 7710 89 21.05 985 20.79\r\n'2022-01-31 14:15:37.566235 b'13 8202 264 45.61 1075 21.06\r\n'2022-01-31 14:15:37.676366 b'14 8312 113 23.02 1077 20.68\r\n'2022-01-31 14:15:37.691931 b'15 8317 110 22.78 1079 20.90\r\n'2022-01-31 14:15:38.469584 b'16 9105 80 20.14 1169 20.87\r\n'2022-01-31 14:15:38.522929 b'17 9158 285 4 [... cut ...] 491 282668 138 25.22 26151 21.06\r\n'2022-01-31 14:20:12.412803 b'492 282857 154 27.02 26153 20.95\r\n'2022-01-31 14:20:12.473869 b'493 282918 390 82.70 26155 21.27\r\n'2022-01-31 14:20:12.491624 b'494 282931 222 37.96 26157 21.06\r\n'2022-01-31 14:20:12.763967 b'495 283208 59 17.27 26247 20.90\r\n'

Can you send the txt file in an email? ***@***.*** ***@***.***>

tony1tf, 2022-01-31:
I did attach the text file to an email, but it seems to have appeared inside the email above - very strange.
I'll try again with an email just to you.
Tony

tony1tf, 2022-01-31:
I've sent it to your MIT email.
Tony