spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #77: Import Data Live

Phoenix1747, 2022-02-15:
Hi there, I'm trying to build CosmicWatch-compatability into my own [gamma spectroscopy software](https://github.com/Open-Gamma-Project/Gamma-MCA) to enable live-plotting without the need for an extra python script. (not for gamma spectroscopy of course, but just for some simple histogram plotting like the current website)

Now I don't have access to a muon detector so I can't test it on my own. What does the serial output from the detector look like when it is sent to the [current web utility](http://www.cosmicwatch.lns.mit.edu/measure)? Is it the same when printing to a file like this?

```text
##########################################################################################
### CosmicWatch: The Desktop Muon Detector
### Questions? saxani@mit.edu
### Comp_date Comp_time Event Ardn_time[ms] ADC[0-1023] SiPM[mV] Deadtime[ms] Temp[C] Name
##########################################################################################
Device ID(s): NCBJ021 
2018-07-10 12:53:32.659892 1 42 608 202.86 74 96.63 NCBJ021
2018-07-10 12:53:33.316180 2 699 108 22.63 78 26.05 NCBJ021
2018-07-10 12:53:33.456973 3 841 152 26.78 82 5.43 NCBJ021
2018-07-10 12:53:33.962563 4 1344 195 33.17 120 12.63 NCBJ021
2018-07-10 12:53:34.458533 5 1841 71 19.06 124 -10.68 NCBJ021
2018-07-10 12:53:34.590431 6 1973 369 73.42 128 40.23 NCBJ021
.....
```

(See [this file](https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/blob/master/Recording_Data/ExamplePlots/Sheilding/bkg_nobox_long.txt))
I'm planning on using the Web Serial API to interface directly with the detector, take it's serial data and the correct data element (which would be number five here - the ADC channel which is most important) and plot it, thus bypassing the need for a local webserver like tornado. This currently only works in Chrome-based browsers though.

spenceraxani, 2022-02-15:
Hi,

The data is continuously sent out through the serial port. That data can be either sent to our website for plotting in realtime, or written to a file, using the import_data.py script. In your case, you should look at how it is written to a file, and have your script monitor that file for changes to update your plots. The data import_data.py script adds the first two columns in the data that you attached. Everything else, is sent directly from the detector.

Thanks,

Spencer

jeroenvantilburg, 2022-04-21:
Hi there,

This is exactly what I have been doing! I have used the Web Serial API to plot live data on a Chrome browser without the need for running a Python script (my students for instance do not have Python on their Chromebooks). My web app can plot both online (live) and offline data, it can plot multiple datasets at the same time, it can modify the histograms and it can find coincidences between two datasets. Have a look at:
  https://jeroenvantilburg.nl/cosmicwatch/

Feel free to use the code for your own purpose in case you haven't implemented it yourself already.

Jeroen

P.S. After working for a year with these detectors I finally found this nice "forum" ;-)

spenceraxani, 2022-05-28:
Hi Jeroen,

Thanks for the link and sorry for the extremely late reply. That looks amazing — great idea! Our web GUI works a bit differently. I’ve run into a handful of people hitting the python wall, either from lack of programing experience or only being familiar with another language. 

Thanks,

Spencer

closed 2022-06-05 by Phoenix1747