spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #80: Replacement for C4 and C16

Airbasti, 2022-04-06:
Hey, 
during building i noticed i accidentally ordered a 10uF 16V X5R0805 instead of the 6.3V one. Since ordering always takes some time from my country i would be interested if i can use another capacitor on the C4 and C16 spots. I ordered the other capacitors in large quantities and could use them. Or maybe the 16V will work but i cannot quite understand its function. 
Thanks in advance!!

Airbasti, 2022-04-06:
Little update, 
in the purchasing list it says 10 uF 16V and in the description of where the components are placed it say 10 uF 6,3V.
Does it matter which i take?

closed 2022-04-08 by Airbasti

spenceraxani, 2022-10-11:
Hi,

Shouldn’t be a problem. The number you are referring to represents a rating for the maximum operating voltage. These large caps (10uF) are used for filtering a 5V line, so anything above 5V should be fine. The 6.3V caps were chosen simply because they were cheaper.

Thanks,

Spencer