spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #81: How do i check if SiPM broke

Airbasti, 2022-04-08:
Hey i just finished building two of the detectors. Such a great project!
Sadly only one works. The other one shows no detections at all. Does anyone have an idea what could be the problem?
The main PCB is working fine. I tested it with the other scintillator. It must be a problem coming from the SiPM PCB. 
I compared all the voltages coming from the SiPM PCB and they are matching the ones from the working SiPM. I even did a test using light and the SiPM seems to react to light because the voltage gets higher coming from the SipM with more light. 
I dismantled the whole board except for the SiPM and rebuild it changing every component apart from the SiPM. I suspect i broke it somehow by tightening the screws to far but i would appreciate it if anyone has another option that could cause these problems. Is there a way to test if my SiPM works? Don't want to replace a working one.
Thank you in advance!!

spenceraxani, 2022-04-08:
Hi,

Glad you enjoyed the project, and it sounds like you are pretty close to having two working detectors. Your troubleshooting was all correct, and the problem likely lies within the connection on the SiPM. A few things to check:
	1. Ensure that the SiPM is correctly orientated (pin 1 to the footprint with the circle).
	2. Add some new solder to the SiPM connections — this is my guess as to your problem. It could be that you simply have a cold solder joint on the SiPM-SiPM PCB.
	3. Make sure that the resistors on the SiPM PCB are soldered correctly. This is unlikely the issue, but add some new solder anyways.
	4. Make sure when you tape it up that it is light-tight. A small light leak can cause the detector to stall. We use single photon detectors, so the smallest light leak can easily saturate the detector. 

It’s unlikely that you broke anything. The above checks should solve the issue. 

Thanks,

Spencer

Airbasti, 2022-04-08:
Okay thank you so much for the quick answer! So i should not see a detection at all if the detector is saturated by light? If i don't add the electrical tape completely it will not work?
And doesen't the fact that the multimeter measures a current on the other side of the SiPM PCB mean that i have soldered the SiPM correctly (I am refering to this test measurement you did here at the 12 minute mark https://www.youtube.com/watch?v=yFgin5wlw4I) ? Or should i check it anyway?

spenceraxani, 2022-04-10:
Hi,

That’s correct. You will not record counts if there is too much light. It really needs to be light tight to ensure that it is working. The multimeter test does not ensure that the orientation is correct, it simply means that you have continuity between the terminals. However, if you have a poor connection, or intermittent, you could potentially still see a current with the multimeter. Since you need to check the SiPM orientation, you should also add some new solder to the SiPM connections and to the resistors on the SiPM PCB before taping it up again.

Thanks,

Spencer

Spirko, 2022-05-05:
You can test the orientation and connection of the SiPM with a multimeter in diode mode.  There are contact points on the bottom of the board (opposite the SiPM).  Just remember that in operation, the SiPM is reverse biased, but when testing with a multimeter, you want it forward biased.  The measured forward bias voltage is around 0.5 V.